Xinyuan-LilyGO/LilyGo-AMOLED-Series #7: Port to esp-idf

krupis, 2024-01-04:
Hey, I have initially struggled a bit to get the T-Display-S3 to work on the esp-idf but I got it to work and made it publicly available at:
https://github.com/krupis/T-Display-S3-esp-idf

Are there drivers available for the AMOLED display on the esp-idf?

lewisxhe, 2024-01-05:
Hi, krupis.
The driver part can be directly copied to esp-idf for use, because the driver method is still the method of calling esp-idf, so it is the same

krupis, 2024-01-05:
When we talk about the driver, are we referring to the following files:

```LilyGo_AMOLED.h```
```LilyGo_AMOLED.cpp```
that are located at https://github.com/Xinyuan-LilyGO/LilyGo-AMOLED-Series/tree/master/src

Are there any other dependancies to get the basic example working on the esp-idf?

lewisxhe, 2024-01-05:
What I said is that the driving method is the same, not that you directly copy the src. Later, I will give a collection example of all AMOLEDs using ESP-IDF drivers.

lewisxhe, 2024-01-31:
I've done most of the basic examples for LilyGo displays, you can find them here
https://github.com/Xinyuan-LilyGO/LilyGo-Display-IDF

closed 2024-02-20 by lewisxhe