Xinyuan-LilyGO/LilyGo-AMOLED-Series #9: Pin diagram needed for T-Display S3 Amoled TOUCH VERSION

richgain, 2024-01-10:
The pin diagram and board picture are for the non-touch S3 Amoled. I have several of these and they work well.
I cannot get any of my code to work on the new Touch version of the T-Display S3 Amoled. I need to check for conflicts between the pins I am using to connect an SD card and the pins used for the touch sensor, but there is no accurate pin diagram available for this board.

lewisxhe, 2024-01-10:
The display and pin headers on both sides are exactly the same as touch and non-touch. The only difference is that touch uses SDA = 3 SCL = 2 IRQ = 21 EN = 38
[image]

richgain, 2024-01-10:
OK, thanks very much.
It appears that my problem was not caused by in incompatibilities, but by this issue: https://github.com/espressif/esp-idf/issues/8521 

SDMMC_FREQ_DEFAULT is set to 20000.
On the S3 Amoled Touch, my code stops with this error:
E (510) sdmmc_common: sdmmc_init_ocr: send_op_cond (1) returned 0x107
E (511) vfs_fat_sdmmc: sdmmc_card_init failed (0x107).
ERROR: File system mount failed!

On the non-touch version it works perfectly.

lewisxhe, 2024-01-11:
Has the problem been solved ?

richgain, 2024-01-11:
No, I haven't been able to get it working yet.

richgain, 2024-01-11:
Changed SDMMC_FREQ_DEFAULT from 20000 to 10000 - did not work.
Reassigned pin mappings for the SD card to:
    // microSD card
    #define SD_SCK 42
    #define SD_MISO 41
    #define SD_MOSI 43
    #define SD_CS 44
Success. This got the files opening in the Serial monitor window, but the screen is still blank.

richgain, 2024-01-11:
Now I am investigating why the example code from LILYGO displays correctly, but my programs don't.

My uses the Arduino_GFX library from @moononournation.
I tried running his PDQgraphicstest on both the Touch and Non-touch versions of T-Display S3-AMOLED.
(https://github.com/moononournation/Arduino_GFX/tree/master/examples/PDQgraphicstest)
The tests run perfectly on the Non-touch version, but the Touch version shows only a black screen.
As before, the Serial monitor window shows that the code is running and delivers the results, but nothing is visible on the screen.

richgain, 2024-01-11:
[image]
NON-Touch

[image]
TOUCH

lewisxhe, 2024-01-12:
As I mentioned above, EN=GPIO38, this is the Pin that controls the power supply to the display screen. Using other libraries, you need to set IO38 to high level separately.

richgain, 2024-01-12:
OK, thanks.
I tried adding the following two lines into my void setup(), but still nothing on the screen yet.
  pinMode(38, OUTPUT);
  digitalWrite(38, HIGH);

Do I also need to add code for any of the other three pins? I assume these are only for the touch sensor.
SDA = 3 SCL = 2 IRQ = 21

lewisxhe, 2024-01-12:
I just tested it and it works fine after setting IO38 to HIGH

lewisxhe, 2024-01-12:
I've added an example so you can test it

richgain, 2024-01-12:
Brilliant! Thanks again. I have it working now.
My mistake was adding the two new lines to the end of my void setup() section. 
As soon as I moved them up to the top everything worked.

lewisxhe, 2024-01-16:
So, the problem is solved, I will close the issue

closed 2024-01-16 by lewisxhe

salasidis, 2024-06-05:
Question - is there a way to optionally make Wire2 = the touch display, and move Wire1 to the UART interface and use that QWIIC connector for that purpose.

I had interference between sensor devices attached to the QWIIC connector and the touch interface. This seems to have been solved by adding mutex exclusions around the I2C access by my sensors, as well as the lv_task_handler()

`   
.
.
.
 do_lv_task_handler();
  }
}

void do_lv_task_handler() {
  xSemaphoreTake(I2C_Mutex, portMAX_DELAY);
  lv_task_handler();
  xSemaphoreGive(I2C_Mutex);
}`

While this works, I have not seen another I2C error in a few days, it does remove a fair amount of available time for the sensors to have access to the I2C port.

dobodu, 2024-06-27:
Hi  Lewisxhe

Once more thank for you help.

Does the T4 S3 Amoled display need also one pin high ? I've tryed to set PIN 38 to high without success.

lewisxhe, 2024-06-28:
@dobodu https://github.com/Xinyuan-LilyGO/LilyGo-AMOLED-Series/blob/299a71b1f79d5f372edfa2c1e0336e0de47b8b87/src/LilyGo_AMOLED.h#L286

dobodu, 2024-06-29:
Oh thanks,
I definitely should buy new glasses since the schematics clearly shows V_EN on GPIO09.

dobodu, 2024-06-29:
Hi lewisxhe,

I've read the initialisation you use for RM690B0.

Would you just indicate me where to find the command explanation, i can't find them in the RM690B0 datasheet

{0xFE, {0x20}, 0x01},           //SET PAGE = OK I find this One
{0x26, {0x0A}, 0x01},           //MIPI OFF and the next one are not described in datasheet ?
{0x24, {0x80}, 0x01},           //SPI write RAM
{0x5A, {0x51}, 0x01},           //! 230918:SWIRE FOR BV6804
{0x5B, {0x2E}, 0x01},           //! 230918:SWIRE FOR BV6804

Thanks a lot