Xinyuan-LilyGO/LilyGo-AMOLED-Series #12: T-Display-S3 AMOLED Touch - Powering from 5v pins not working, and serial not working

JackDaviesDesign, 2024-01-19:
Hi,

I am trying to power the board from 5v using the VBUS pin (5v) and GND, however the board does not turn on. It works fine using USBC, and I am turning GPIO38 HIGH on setup to enable the screen. 

Also the serial pins, 44-Rx & 43-Tx do not work, also using USB serial does not work, how can I enable the serial communication?

Thank you

JackDaviesDesign, 2024-01-19:
I actually got the serial working by changing to Serial1 and explicitly defining the Rx & Tx pins.

`Serial1.begin(115200, SERIAL_8N1, 44, 43);`

Still cant power the device from the 5v pins however.

JackDaviesDesign, 2024-01-19:
Okay I guess somehow the failing UART connection was causing the power not to work? The board is working on 5v now too. Hope this helps someone else!

closed 2024-01-19 by JackDaviesDesign

lewisxhe, 2024-01-20:
https://github.com/Xinyuan-LilyGO/LilyGo-AMOLED-Series#7%EF%B8%8F%E2%83%A3-faq