Xinyuan-LilyGO/LilyGo-AMOLED-Series #13: GPIO connector is too small and more examples are required

Arielhh, 2024-01-21:
lilygo development team,
for the T4S3 How do you expect us to connect to such small pins? The provided female connector should be attached to a dedicated PCN to which we can solder the wires. providing only the connector does not solve the problem.
Also, more examples are required. it is too complicated now to understand how to work with that board.

lewisxhe, 2024-01-21:
The remaining GPIO number of esp32s3 cannot place a larger connector (2.54mm) on T4S3, so using 1.27mm spacing is the most reasonable choice.

Regarding examples, what do you need?

Arielhh, 2024-01-21:
Yes, but it is very complicated to solder wires to the female connector. It was better to attache the supplied female connector to a small PCB that allows wires to be connected more easily

lewisxhe, 2024-01-21:
Do you mean to use a 1.27mm to 2.54mm PCB and insert it as a shield on the T4S3?

lewisxhe, 2024-01-21:
Okay, I will give your feedback. I think there should be no problem with this. I think it is more appropriate to use a PCB shield than a wire.

Arielhh, 2024-01-21:
I agree. Many thanks.

Arielhh, 2024-01-25:
Can you give a simple example using TFT_eSPI (not with Sprite)?

lewisxhe, 2024-01-26:
Hi.
1.Which board are you using? 
 2. You said that after writing most of the examples, the screen is still black. Can you point out which examples?  
3. Examples other than the TFT_eSPI wizard cannot be used because TFT_eSP does not support QSPI interface screens.

Arielhh, 2024-01-26:
I am using the T4-S3. After a long time trying I managed to upload all the example codes but they seems to be too complicated.
I understand now that TFT_eSPI doesn't support QSPI interface screens. Can you provide a simple example on how to upload an image to the screen or drawing one circle? 
Is that the optimal way of showing graphics on this board?

lewisxhe, 2024-01-27:
Currently, I think the most appropriate way is to use the lvgl graphics framework. If you are not familiar with it, you can check out the lvgl documentation, which is very easy to use.

Of course, you can also use the TFT_eSP wizard.

lewisxhe, 2024-02-26:
T-Display-Long-Shield

[image]

T-Display AMOLED No plans for now

Arielhh, 2024-02-26:
is it possible to purchase the T-Display-Long-Shield?

lewisxhe, 2024-02-27:
> Looks great :) is it possible to purchase the T-Display-Long-Shield? BTW: the pins in this shield are not aligned with the actual connector of the T-Display Long.

 ? Not right? Where are you referring to?

[image]

lewisxhe, 2024-02-27:
[image]
T4-S3 shield is also completed today

Arielhh, 2024-02-27:
> > Looks great :) is it possible to purchase the T-Display-Long-Shield? BTW: the pins in this shield are not aligned with the actual connector of the T-Display Long.
> 
> ? Not right? Where are you referring to?

Yes, now it is ok. From the last photo, it was not possible to see the male pins of the connector. 
When it will be possible to buy them?

lewisxhe, 2024-02-27:
Still need to wait for some time

Arielhh, 2024-02-27:
> image.png (view on web)
> T4-S3 shield is also completed today

In the long version shouldn’t it be a female connector instead of the male
shows in the picture?

rhcp4life, 2024-03-21:
Any updates on the long shield?

I just ordered a IDC ribbon connector like this: 
uxcell IDC Wire Flat Ribbon Cable FC/FC Female Connector A-Type 30Pins 1.27mm Pitch 20cm Length
https://www.amazon.com/dp/B07SN9CYCY?ref=ppx_yo2ov_dt_b_product_details&th=1

lewisxhe, 2024-04-03:
Production has been arranged and will be available for purchase soon.

Arielhh, 2024-04-03:
Great news, thanks.

closed 2024-05-18 by github-actions[bot]