Xinyuan-LilyGO/LilyGo-AMOLED-Series #18: Issues After Update (Arduino IDE)

cairogomes, 2024-01-30:
After the update (1.0.6), even downgrading Lvgl to 8.3.x, the screen displays a partial broken image, a small rectangle in the upper left corner, the rest of the screen is black.

mauromorello, 2024-01-30:
From my side after after update everything was messed up...

cairogomes, 2024-01-30:
> From my side after after update everything was messed up...

Before update everything was working perfectly and now everything is really messed up. Let's see if they release a fix.

lewisxhe, 2024-01-31:
Yeah ? I'll check it out, can you tell me which example you ran? Currently I use platformio and everything works fine

lewisxhe, 2024-01-31:
Should I use the TFT_eSPI_Sprite example?

lewisxhe, 2024-01-31:
I guess you are using TFT_eSPI_Sprite, because the rotation function was added and the width and height were changed, which caused the original example to run incorrectly. Just swap the width and height.
Please confirm
https://github.com/Xinyuan-LilyGO/LilyGo-AMOLED-Series/commit/cf75c441e97717d89fdece73aaef0c33eb2d86dc

mauromorello, 2024-01-31:
I had an example (v8.3) customized with my stuff.
After updating old Font files (obtained from Lvgl site) are not working anymore (cache, last part of the file).
This evening I will be more specific

lewisxhe, 2024-01-31:
Hi.@mauromorello 
You can see my changes from the commit information. I only changed the full-screen refresh to partial refresh. What does this have to do with the font?  I look forward to your detailed explanation.

cairogomes, 2024-01-31:
Now this is working for me:

#define LILYGO_TDISPLAY_AMOLED_SERIES

#include "esp_arduino_version.h"
#include <LilyGo_AMOLED.h>
#include <TFT_eSPI.h>

TFT_eSPI tft = TFT_eSPI();
TFT_eSprite spr = TFT_eSprite(&tft);
LilyGo_Class amoled;

#define WIDTH amoled.width()
#define HEIGHT amoled.height()

void setup() {
  bool rslt = false;
  Serial.begin(115200);
  rslt = amoled.begin();
  if (!rslt) {
    while (1) {
      Serial.println("There is a problem with the device!~");
      delay(1000);
    }
  }
  spr.createSprite(WIDTH, HEIGHT);
  spr.setSwapBytes(1);
}

void loop() {
  amoled.pushColors(0, 0, WIDTH, HEIGHT, (uint16_t *)spr.getPointer());
  spr.fillSprite(TFT_BLUE);
  spr.setTextSize(2);
  spr.setTextColor(TFT_WHITE);
  spr.setTextWrap(false);
  spr.setTextDatum(4);
  spr.drawString("Hello World! :)", WIDTH / 2, HEIGHT / 2, 4);
}

mauromorello, 2024-01-31:
> Hi.@mauromorello You can see my changes from the commit information. I only changed the full-screen refresh to partial refresh. What does this have to do with the font? I look forward to your detailed explanation.

You are right, obviously. I think my problem is related to lvgl 9 and my code, not because your update.

closed 2024-03-16 by github-actions[bot]