Xinyuan-LilyGO/LilyGo-AMOLED-Series #21: Screen tearing on the T-Display S3 AMOLED V1.0

diegomacario, 2024-02-19:
Hi! I wrote a little menu for my T-Display S3 AMOLED V1.0. It works great! You can see it in this video:

https://github.com/Xinyuan-LilyGO/LilyGo-AMOLED-Series/assets/8304271/14d6bcd9-5f91-4f10-b0fb-3595074df25f

What I wanted to ask here is if there is a way to eliminate the tearing I see when I update the boxes in the screen. You can see the tearing in the video above, but it's easier to see it in these screenshots:

[image]
[image]

I'm using sprites and only updating the parts of the screen that change, but I still see tearing.

I also see the same problem in this example from Volos Projects that I modified so that the dial boxes periodically turn red:

[image]
[image]
[image]

Is there anything I can do about this? Thank you for any advice!

lewisxhe, 2024-02-20:
If the refresh speed is too fast, tearing can be clearly seen. Reduce the refresh speed, or use partial refresh, but partial refresh is limited to only allow even coordinate values, or use lvgl

diegomacario, 2024-02-20:
Thanks for the reply @lewisxhe! When you say partial refresh, do you mean only updating the part of the screen that changed like this?

```
amoled->pushColors(boxXPos, boxYPos, boxWidth, boxHeight, (uint16_t *)boxSprite.getPointer());
```

Because that's what I'm currently doing, and I only call `amoled->pushColors` when the button is pressed. So in my `update` function I'm not calling it all the time. Is that what you mean with reducing the refresh speed i.e. not calling `amoled->pushColors` every frame?

Thanks for any information!

lewisxhe, 2024-02-21:
Yes, I have no other better way to prevent screen tearing

diegomacario, 2024-02-21:
Alright, I guess I just have to avoid having large parts of my user interface change colors suddenly.

The displays seems to have support for some form of V-Sync:

```
    // {0x44, {0x01, 0x66},        0x02}, //Set_Tear_Scanline
    // {0x35, {0x00},        0x00}, //TE ON
    // {0x34, {0x00},        0x00}, //TE OFF
    // {0x36, {0x00},        0x01}, //Scan Direction Control
```

But I'm not sure if that would work with QSPI.

lewisxhe, 2024-02-26:
At present, I only know that RGB panels support TE configuration, but QSPI currently does not see how to configure it.

diegomacario, 2024-04-06:
A fix for this bug was developed in this thread: https://github.com/Xinyuan-LilyGO/T-Display-S3-AMOLED/issues/2

So I will close this one now. Thank you!

closed 2024-04-06 by diegomacario