Xinyuan-LilyGO/LilyGo-AMOLED-Series #23: Top part screen keeps flickering on my T-Display S3 Amoled

hurnyy, 2024-03-10:
Top part screen keeps flickering on my T-Display S3 Amoled.
Video:
https://github.com/Xinyuan-LilyGO/LilyGo-AMOLED-Series/assets/94316699/26af6291-0121-4a26-a5f9-0f5b04f59b8a
Image:
[image]

I was using this video as a tutorial: https://www.youtube.com/watch?v=Waq2xI-t948

With the sketch example from: https://github.com/VolosR/TAmoledTouch/tree/main/Examples

There are three examples and only the TestTouch works. The rest has the same issues of flickering at the top of the screen.

I wanted to try the Binary Converter but it doesn't work as shown in the video.

Any ideas what is wrong?

lewisxhe, 2024-03-10:
It looks like it is damaged. Try writing this binary file to see if it has the same effect. If it still displays abnormality after writing, contact the seller for replacement.

https://github.com/Xinyuan-LilyGO/LilyGo-AMOLED-Series/tree/master/firmware

hurnyy, 2024-03-10:
Yes, I done that as well to be sure and its still the same. I guess its a return then.

lewisxhe, 2024-04-03:
It has been determined that this is a hardware problem, please contact your dealer

closed 2024-04-03 by lewisxhe