Xinyuan-LilyGO/LilyGo-AMOLED-Series #24: T-Display Amoled Touch version 1.91: Can't get brightness to work

PlastiBots, 2024-03-18:
I have the [T-Display Amoled Touch version 1.91](https://www.lilygo.cc/en-ca/products/t-display-s3-amoled)" and cannot get amoled.setBrightness(); to work.  Tried other ways such as Analog_Write, LCD_WRITE etc.. Nada.   What am I doing wrong?  Is there a proper working example someone can share for this unit?

Simple example:

In Setup:
`amoled.begin();   `

Then in code:
```
  amoled.setBrightness(2);
  delay(1500);
  amoled.setBrightness(8);
  delay(1500);
```

I've also tried values from 1-255 as there is no documentation on this. is it PWM, or other values?

lewisxhe, 2024-03-19:
Range 0 ~ 255, example has been added. Brightness control is to write commands directly to the screen instead of using control methods such as pwm.

PlastiBots, 2024-03-19:
Thanks.  However, an LVGL example is not very helpful as I'm not an LVGL user and just trying to get this working would be a steep learning curve.  Can you provide an example just using Arduino pls?  If it's as simple as my example above, and using amoled.setBrightness(50); or amoled.setBrightness(200);  (as an example), this did not work for me.

PlastiBots, 2024-03-21:
Below is a simple sample that I cannot get to work.  It draws the red circle, but brightness does not change. Note I am using the Sprite library from Matt (link within code).   FYI, I have confirmed the Sprite library works with Arduino GFX with other projects.  

`#include <Arduino.h>
#include <LilyGo_AMOLED.h>
#include "Wire.h"
#include <Arduino_GFX_Library.h> // https://github.com/moononournation/Arduino_GFX.git
//#include <DS3231.h>              //RTC time for DS3231
//#include <Adafruit_Sensor.h>     //ADXL345 gyro / accell
//#include <Adafruit_ADXL345_U.h>  //ADXL345 gyro / accell

// Image Sprites - see /lib folder
#include "arduino_sprite.h"                    //https://github.com/h5n1xp/Arduino_Sprite/

#define BYTE_SWAP true                              // Sprites - needed for LCD Image generator tool and how it outputs img data

#define GFX_DEV_DEVICE LILYGO_T_DISPLAY_S3_AMOLED
Arduino_DataBus *bus = new Arduino_ESP32QSPI(
    6 /* cs */, 47 /* sck */, 18 /* d0 */, 7 /* d1 */, 48 /* d2 */, 5 /* d3 */);
Arduino_GFX *gfx = new Arduino_RM67162(bus, 17 /* RST */, 0 /* rotation */);

Arduino_GFX *gfx2;  
Arduino_Canvas *digiClockCanvas; // Canvas for font sprites

LilyGo_Class amoled;

void setup()
{
 //The difference between the touch and non-touch versions is that the display power supply of the touch version is controlled by IO38/
 pinMode(38, OUTPUT);
 digitalWrite(38, OUTPUT);
 bool rslt = false;
  
  Serial.begin(115200);

  //amoled.begin();    // Automatically determine the access device  - this is //amoled.beginAMOLED_191();  Seems this is needed as well
  rslt = amoled.beginAMOLED_191();  
  if (!rslt) {Serial.println("AMOLED BEGIN failed!"); }

 
     
  
  // Init Display
  if (!gfx->begin())
  {
      Serial.println("gfx->begin() failed!");
  }
  //Serial.println("1?"); 
  //gfx->setRotation(3);
  //Serial.println("2?"); 

  //gfx2 = new Arduino_Canvas(480, 480, gfx, 0, 0); // for Sprites
  gfx2 = new Arduino_Canvas(240, 536, gfx, 0, 0); // for Sprites
  gfx2->begin(GFX_SKIP_OUTPUT_BEGIN);             // Added the GFX_SKIP_OUTPUT_BEGIN so the Canvas class doesn’t try and initialise the display
  gfx2->setRotation(3);
  
  //gfx2->setRotation(0);
  //Serial.println("3?"); 

  gfx2->fillScreen(BLACK);
    

} // end Setup

void loop()
{

for (int i=0; i<255; i++)
{
  gfx2->fillCircle(130,130, 40, BLACK);
  gfx2->flush();
  amoled.setBrightness(i);  
  gfx2->fillCircle(130,130, 40, RED);
  gfx2->flush();
  delay(50);
  
}
}
`

lewisxhe, 2024-03-21:
If you use it as above, adjusting the brightness will of course not work, because gfx also calls SPI, they belong to two categories completely and cannot be used together. I have modified it slightly and it can be used now.

PlastiBots, 2024-03-21:
Hi. Thanks for the quick response, and it works. However, if I am not able to use the LilyGo library, I can't use the home button callback and getPoint for touch.    Can you help with the necessary code to enable those pls?

```
amoled.setHomeButtonCallback([](void *ptr) {
        Serial.println("Home key pressed!");
        static uint32_t checkMs = 0;
        checkMs = millis() + 200;
        gfx2->fillRoundRect(8,0,180,30,4,BLACK);
        gfx2->setCursor(20, 22);
        gfx2->print("H: App:  "); 
        gfx2->print(appScreen); 
```

`bool touched = amoled.getPoint(&x, &y);`

lewisxhe, 2024-03-22:
Of course, you can use touch. The source codes are all in the src directory. You can change them freely after downloading. I've modified the code to be touch compatible, hope this is your last question.

PlastiBots, 2024-03-22:
Thanks for this.  However, when I try to integrate that code it causes IIC failures.   Issue is I have 2 other IIC devices and use 
`  Wire.begin(IIC_SDA_PIN, IIC_SCL_PIN, (uint32_t)400000);`  (which as worked so far), and I believe that adding  `touch.begin(Wire, CST816_SLAVE_ADDRESS, 3, 2);` causes it to fail.  Given my pins are not 3&2, I also tried to use the same IIC pins `touch.begin(Wire, CST816_SLAVE_ADDRESS, IIC_SDA_PIN, IIC_SCL_PIN);`  but I think the conflict is calling .begin twice.       I also tried to use touch.begin(); but that fails too as it requires parameters.

[599235][E][Wire.cpp:513] requestFrom(): i2cRead returned Error -1

lewisxhe, 2024-03-22:
two methods
1. The external sensor uses the same SDA and SCL as the touch
2.Change touch initialization to
```
touch.begin(Wire1, CST816_SLAVE_ADDRESS, 3, 2);
```

PlastiBots, 2024-03-22:
This is working now..  I decided on #2 (keep the touch on 3/2) and put my other IIC devices on another channel.  Thanks for your help!

closed 2024-03-22 by PlastiBots