Xinyuan-LilyGO/LilyGo-AMOLED-Series #25: T4-S3 3D PCB model - incomplete

pike2k, 2024-03-18:
The file here: dimensions/T4-S3-AMOLED-PCB-3D.7z is lacking many dimensions and features. Can you provide the full 3D model?

lewisxhe, 2024-03-19:
Already updated to the latest version of 3D, please note it only has the parts you need for the shell design part

pike2k, 2024-03-25:
I need more dimensions, since I plan to design an 'Expansion board' and it would help a lot to know for instance where the 2x15 pin connector is located on the PCB!

lewisxhe, 2024-03-27:
[image]
Isn't this up there?

pike2k, 2024-03-27:
[image]
I use FreeCAD 0.21.2 and this is all I get from T4-S3-AMOLED-PCB-3D.step

lewisxhe, 2024-03-27:
All right . I put it in the wrong directory. it's OK now

Serhii-Khramenkov, 2024-08-06:
> ![image](https://private-user-images.githubusercontent.com/22990954/317092748-e7cce8bd-d448-4d71-9734-39fc582c6ebd.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3 [...]

It's 2x16 pins connector on the photo, the same is in 3D model. There is 2x15 connectors on my version of display. Could You please clarify the situation ? Are there different versions ?

pike2k, 2024-09-23:
I am designing a breadboard adapter for this product and I just got a big scare. In the Step file you provide, the pinheader is 2x16 pins. In reality it is 2x15 pins. Could you please fix it and position it where it should be on the PCB?

lewisxhe, 2024-09-24:
Already updated

closed 2024-09-24 by pike2k