Xinyuan-LilyGO/LilyGo-AMOLED-Series #26: Wifi is very weak on t4 S3

isthaison, 2024-03-20:
T4 s3 and wifi signal is too bad. with signal capabilities that it can achieve

```
#include "WiFi.h"
void setup()
{
    Serial.begin(115200);
    bool err = false;

    WiFi.mode(WIFI_STA);

    Serial.println();
    Serial.println("******************************************************");
 

    WiFi.begin("SO95", "12345678");

    while (WiFi.status() != WL_CONNECTED)
    {
        delay(500);
        Serial.print(".");
    }

    Serial.println("WiFi Connected.");

    Serial.print("IP Address: ");
    Serial.println(WiFi.localIP());
    Serial.print("RSSI: ");
    Serial.println(WiFi.RSSI());
}

void loop()
{
    // put your main code here, to run repeatedly:
    Serial.print("RSSI: ");
    Serial.println(WiFi.RSSI());
   

    delay(1000);
}
```
RSSI: -64 while both devices are almost in contact with each other

isthaison, 2024-03-20:
https://github.com/Xinyuan-LilyGO/LilyGo-AMOLED-Series/assets/22739578/3f33fe45-2b22-4aa1-b6a2-96d2b9881e1f

lewisxhe, 2024-03-21:
Take a photo of the back of the board and the wifi antenna part to see

isthaison, 2024-03-21:
[image]

lewisxhe, 2024-03-21:
[image]
There should be two resistors here, right?

isthaison, 2024-03-21:
I just kept the board original for testing. You can test directly with your version.

lewisxhe, 2024-03-21:
The test results here will not be very bad. It is about 6 meters away from the router. There is a glass block in the middle. The signal is around -60. Can you remove the PCB on the back for testing?

isthaison, 2024-03-21:
[image]
Can you help me get in touch with Skype, I'll video call you directly.

lewisxhe, 2024-03-21:
sorry, we do not have that . This kind of problem cannot be solved remotely.

isthaison, 2024-03-21:
https://github.com/Xinyuan-LilyGO/LilyGo-AMOLED-Series/assets/22739578/2c917b33-a041-49a2-a665-ffe5a069b8f2

isthaison, 2024-03-21:
I have found a problem on the board

isthaison, 2024-03-21:
[image][image][image]
#I had replace and repaired all antennas  achieved 7x rssi instead of 9x before

isthaison, 2024-03-21:
For the signal line from the expansion antenna transmitted to the esp32, you installed a missing resistor. It caused the expansion antenna to not work. This is a serious error if the board is placed in a metal shell. then the built-in antennas on the board will not well work. In addition, the antennas integrated on the board have poor signal capture performance. You should consult about hardware design from the creator of the esp32 chip. How did they design the esp32 module?[image]

lewisxhe, 2024-03-22:
Maybe you are a special case, the signal in my hand is not that bad.

isthaison, 2024-03-23:
I have a suggestion, you try to operate it continuously for 24 hours then check the results again

lewisxhe, 2024-03-29:
After three days of testing, the signal was kept at about - 65 dBM when the distance from the router was about 6 to 7 meters. Maybe you are an exception.

lewisxhe, 2024-04-03:
If you have any other questions later, please contact your dealer.

closed 2024-04-03 by lewisxhe