Xinyuan-LilyGO/LilyGo-AMOLED-Series #28: T4-S3 Portrait vs. Landscape

GSGDaren, 2024-03-30:
Has anyone been successful in rotating their display on the T4-S3 using LVGL?  I saw a post related to T3 that stated: 

> Changing these two lines will change the screen orientation
> 
>     esp_lcd_panel_mirror(panel_handle, false, true);
>      esp_lcd_panel_swap_xy(panel_handle, true);

But I see these values nowhere in the AMOLED codebase.

GSGDaren, 2024-03-31:
I've tried adding 

>     disp_drv.sw_rotate = 1;
>     disp_drv.rotated = LV_DISP_ROT_90;

to the _beginLvglHelper_ function, but this causes screen tearing on the bottom portion of the rotated screen.  Variations of the two property settings resulted in either no display at all or non-rotation.

LVGL forums indicate software rotation is not compatible with FULL_REFRESH.  So, is there hardware supported screen rotation on the T4-S3?

GSGDaren, 2024-03-31:
Found it:

> board.setRotation(45);

Why does 45 work?  ¯\_(ツ)_/¯  Seems that 90 Flips the display lengthwise and 180 is back to regular.

lewisxhe, 2024-04-03:
Have you seen this example?

https://github.com/Xinyuan-LilyGO/LilyGo-AMOLED-Series/blob/633354f207f36c1ff6fd4d06211532731b8238c0/examples/LVGL_Rotation/LVGL_Rotation.ino#L28

GSGDaren, 2024-04-04:
Thanks Lewis, that seems like a much more appropriate method to use.

lewisxhe, 2024-04-08:
If there is anything else, please reopen the issue

closed 2024-04-08 by lewisxhe