Xinyuan-LilyGO/LilyGo-AMOLED-Series #30: Can't get Lvgl_Images.ino working on the T4-S3

richgain, 2024-04-07:
I can get examples/Lvgl_Images/lvgl_images.ino to work on my T-Display-Amoled S3 Touch, but not on the T4-S3.
I had to modify the code because I kept running out of memory. I can load six images into the S3 Touch and four images into the T4-S3.
When I do this, the S3 displays the six images in a loop when touch the action button, but the T4-S3 only shows a black screen.

I can see in the void setup() section it says:
    // Only 1.91 Inch AMOLED board support

I assumed that this just referred to the action button, but I can't get the T4 to display any images at all. Is there any way to get this code working on the T4-S3?

Eventually, I am trying to get the T4-S3 to display a series of JPEG files stored on a microSD card in the SD slot.

lewisxhe, 2024-04-08:
I have no problem testing it and it runs fine except for the error described.

lewisxhe, 2024-04-08:
Here is an example of an SD card displaying a picture. Regarding ArduinoIDE, you can only reduce pictures and place them in Flash, or use platformio
[ImageDecoder](https://github.com/Xinyuan-LilyGO/LilyGo-AMOLED-Series/blob/master/examples/lvgl/third_party/ImageDecoder/ImageDecoder.ino)

lewisxhe, 2024-04-08:
I've added another example showing images from the SD card

richgain, 2024-04-08:
Thanks Lewis. 
Both of these are really helpful.

closed 2024-04-08 by richgain