Xinyuan-LilyGO/LilyGo-AMOLED-Series #31: amoled.isBatteryConnect() always returns 0 - T-Display S3 AMOLED 1.91 v2 touch [H681]

johnwcha, 2024-04-09:
all sample code compiles and runs fine
running fine on a 1100 mAh Lipo battery
amoled.getBattVoltage() seems to work fine either via usb or on battery

However, trying to detect if device is running on battery power, amoled.isBatteryConnect() would always return 0
amoled.isCharging() always returns 0 even though the battery is charging

would appreciate any help. Thanks

StevenLCox, 2024-04-09:
Johnwcha,
I don't think that function will work because the 1.91 touch does not have
a battery power management unit/PMU like the 1.47 lite version.
SteveC

johnwcha, 2024-04-09:
I see. Thanks, Steven. 

On the website, it says The module is equipped with a built-in battery management system. Is there any function to work with battery status/charging?

StevenLCox, 2024-04-09:
Take a look at the 1.91" version schematic. It only has a TP4065 based charging circuit. It's a dumb charger with no control/status interface to the processor. So no status feedback is possible, The ESP32 can only read the BAT_VOLT voltage on GPIO4. The 1.47" version has a smart interface.

johnwcha, 2024-04-09:
Thanks. You're super !!!

StevenLCox, 2024-04-09:
Surprised it didn't throw an error. Those functions should only work for BOARD_AMOLED_147 and BOARD_AMOLED_2417 versions.
No problem, glad it helped!

StevenLCox, 2024-04-09:
Forgot to mention that amoled.getBattVoltage() does work for 1.91". It just uses the ESP32 S3 built in ADC. Gives millivolts.

closed 2024-05-25 by github-actions[bot]