Xinyuan-LilyGO/LilyGo-AMOLED-Series #32: Confusion over T-Display S3 AMOLED screen settings

samarkh, 2024-04-13:
I have just taken delivery of two T-Display S3 AMOLED boards. I cant get any display working using PlatformIO, My boards only have two buttons, BOOT and RESET. Which driver, and pins do I need?

lewisxhe, 2024-04-25:
The version you received should be a touch version, written directly to the Factory, or are other examples not shown? Then you should use the [firmware](https://github.com/Xinyuan-LilyGO/LilyGo-AMOLED-Series/tree/master/firmware) to test whether there is any problem with the board.

dobodu, 2024-04-30:
I'm also facing issue concerning the S3 amoled touch display initialisation.

It's working fine with your repository but when i try a Micropython version, based on the non touch version, i cannot manage to get the screen working.

Would you have a look there and tell us if we do something wrong ?

https://github.com/nspsck/RM67162_Micropython_QSPI/issues/2

Thank a lot for any help

closed 2024-06-14 by github-actions[bot]