Xinyuan-LilyGO/LilyGo-AMOLED-Series #33: Watchdog gets triggered after a short while

Daanoz, 2024-04-15:
Hello, I'm running a slightly modified factory example on the T4-S3 board. But I keep getting resets due to the watchdog triggers:

```
E (419818) task_wdt: Task watchdog got triggered. The following tasks did not reset the watchdog in time:
E (419818) task_wdt:  - loopTask (CPU 1)
E (419818) task_wdt: Tasks currently running:
E (419818) task_wdt: CPU 0: IDLE
E (419818) task_wdt: CPU 1: IDLE
E (419818) task_wdt: Aborting.

abort() was called at PC 0x42054edc on core 0
```

I tried to keep the modifications as minimal as possible, so I only modified the `updateWeatherTask` function:
- to poll the api every 5 seconds.
- to use a local network json api over http (using `WiFiClient.h`) to read and update a temperature value.

I ran into this issue on the `lvgl` repo, could it be related? https://github.com/lvgl/lvgl/issues/2972

Daanoz, 2024-05-16:
It still occurs...

The-Young-Maker, 2024-06-18:
You should share the code you modified. Otherwise nobody will be able to help find the problem.

closed 2024-08-03 by github-actions[bot]