Xinyuan-LilyGO/LilyGo-AMOLED-Series #34: Two finger touch patterns cause loss of touch response

Minorplanets76, 2024-04-20:
I have found that two finger gestures such as pinch cause the touch functionality to stop working.  I confirmed this with LVGL_rotation example as supplied, but has been an issue with my own ui's I have created.  Once a two finger event has happened there seems to be no way recover other than reboot.  Needs to ignore unsupported gestures or have a way to break from the scenario.

lewisxhe, 2024-04-22:
I used the LVGL_Rotation example and ran it on T4 -S3, AMOLED - 1.91 and did not find this problem

Minorplanets76, 2024-04-24:
Hi, thanks for the prompt response.  I tried with a brand new board and found the same issue.  I have the T-Display S3 board with touch.  To replicate issue, I load rotation example, push the set rotation button to confirm touch is working.  Touch with one finger at top of screen (not touching buttons) and at same time swipe upwards with thumb.  Buttons are no longer responsive and have reboot.  I can do a video to demonstrate if you'd like.
[image]
It seems like it enters a state or event from which it never exits.

lewisxhe, 2024-04-25:
I can't test the problem, it would be better if you could attach a video.

Minorplanets76, 2024-04-28:
https://github.com/Xinyuan-LilyGO/LilyGo-AMOLED-Series/assets/107326485/3bc9581c-77b1-4f87-ac0e-724f4cbc7ba7

lewisxhe, 2024-04-29:
From the video content, it seems that lvgl is frozen, but the touch response is still normal. It can be ruled out that the problem is caused by touch. I need time to investigate.

lewisxhe, 2024-04-29:
The touch driver has been updated. You can test it. I did not find this problem in my test here. You should be using platformio, right?

vpsmedeiros, 2024-04-29:
I had the same issue with the touch screen becoming unresponsive, but I did not know why or when it would happen.  I just replicated the issue by pinching the screen while pressing a touch screen button.

Initial test with your latest revision seems to have fixed my issue.  Thanks!

Minorplanets76, 2024-04-30:
Yes that appears to have fixed the issue.  Thanks

closed 2024-04-30 by Minorplanets76