Xinyuan-LilyGO/LilyGo-AMOLED-Series #36: Location of external psram component

amir1387aht, 2024-04-26:
Can somebody show me which component is psram?

StevenLCox, 2024-04-27:
The ESP32S3 R8 has the 8MB PSRAM built into it's package. Basically it's a multichip module. So no external RAM on the LilyGo PCB is needed. Take a look at the ESP32S3 dataset page 10. There is an external flash chip on the PCB.
-Steve

amir1387aht, 2024-04-28:
> The ESP32S3 R8 has the 8MB PSRAM built into it's package. Basically it's a multichip module. So no external RAM on the LilyGo PCB is needed. Take a look at the ESP32S3 dataset page 10. There is an external flash chip on the PCB. -Steve

Thank you!

amir1387aht, 2024-04-28:
> The ESP32S3 R8 has the 8MB PSRAM built into it's package. Basically it's a multichip module. So no external RAM on the LilyGo PCB is needed. Take a look at the ESP32S3 dataset page 10. There is an external flash chip on the PCB. -Steve

And a question, it is possible to add more psram?

StevenLCox, 2024-04-28:
I guess there are two ways to add more RAM. Looking at the ESP32S3 datasheet there is the ESP32-S3R16V version that has 16MB of in-package PSRAM. You could also make your own board with an SPI or QPI PSRAM chip like this [](https://www.adafruit.com/product/4677). 

Also maybe something like this: [https://machdyne.com/product/qqspi-psram32/](url), [https://github.com/machdyne/qqspi](url). (I haven't looked at these in any detail but am guessing that the ESP32S3 could support them. Not sure how they would fit into the memory map.)

amir1387aht, 2024-04-29:
> I guess there are two ways to add more RAM. Looking at the ESP32S3 datasheet there is the ESP32-S3R16V version that has 16MB of in-package PSRAM. You could also make your own board with an SPI or [...]

Thanks!

closed 2024-06-13 by github-actions[bot]