Xinyuan-LilyGO/LilyGo-AMOLED-Series #40: Using the IPEX antenna connector

radioralph, 2024-05-13:
I wish to use an external antenna with the T-Display S3 AMOLED. Is the IPEX socket connected, or do resistors/capacitors need to be inserted or removed?

isthaison, 2024-05-13:
@
[image]
I also saw this same thing 2 months ago

radioralph, 2024-05-13:
Thank you @isthaison for your observation regarding the antenna connector on the T4-S3 board. The schematic diagram for the T-Display-S3 suggests that some components will need to be added to enable the IPEX connector, and some components will need to be removed to disable the on-board 3D antenna. The resistors noted are 0R, but the cap values (if required) are not mentioned.

If I am correct in my understanding, the question becomes: what are the recommended values and placement locations of the added/removed components?

Best regards,

isthaison, 2024-05-13:
[image]

radioralph, 2024-05-15:
Thank you @isthaison - that looks straightforward. I will let you know how it goes.
Many thanks again.

isthaison, 2024-05-29:
@radioralph  Is it Ok?

closed 2024-07-13 by github-actions[bot]