Xinyuan-LilyGO/LilyGo-AMOLED-Series #41: T-Display v1 vs v2

roblatour, 2024-05-21:
I wrote this program for the original T-Display S3 AMOLED:
webpage: https://rlatour.com/youtubesubscribercounter/
source: https://github.com/roblatour/youtubescribercounter?tab=readme-ov-file

However, now I have someone who has raised an issue saying the display does not work with the newer T-Display S3 AMLOLED v2.

Any reason why this would be?

lewisxhe, 2024-05-24:
https://github.com/Xinyuan-LilyGO/LilyGo-AMOLED-Series/issues/9#issuecomment-1888466187

roblatour, 2024-06-01:
Thanks, the person who reported this issue to me is saying that did not resolve the problem.

https://github.com/roblatour/youtubescribercounter/issues/1

Unfortunately, while I have a few T-DISPLAY S3 and T-DISPLAY S3 AMOLED v1 boards, I don't have one of the newer v2 boards to test with to confirm myself.

lewisxhe, 2024-06-11:
[image]

[image]

[image]

roblatour, 2024-06-11:
Thank you, I have asked the person that originally posted the question to confirm it is working for them too

roblatour, 2024-06-11:
Is there a way to query the board and have it report if it is a AMOLED v1 or v2?

asg02fever, 2024-06-12:
https://youtu.be/zR0kbOCVvRY

i made a complete fresh install on the firmware change nothing exepte the stuff u said on the last post all change made are in the video

asg02fever, 2024-06-12:
this worked thanks a lot guys am i allowed to post this ported code for the v2 board on my gith page  ?