Xinyuan-LilyGO/LilyGo-AMOLED-Series #42: How to use with LVGL from SquareLine Studio?

The-Young-Maker, 2024-06-02:
Hi, would it be possible to provide a template code to use with LVGL so that we only need to include the UI file from SquareLine Studio to make it work? As I have never been able to make it work because of the special library needed to make the display work (it does not work with TFT_eSPI).

vpsmedeiros, 2024-06-03:
I've been using Squareline studio to make the UI on my board  with LVGL and this is the pertinent code I am using to link the UI to my code...And I only use Export Files. Export path settings for my project is '/display' folder where I keep all the Squareline Studio UI files.  LVGL include path '/libdeps/lvgl/lvgl.h'

`#include <Arduino.h>
#include <LilyGo_AMOLED.h>
#include <LV_Helper.h>
#include "display/ui.h" //path to squareline studio export
#include "display/lvgl.h" //path to squareline studio export

extern LilyGo_Class amoled;

void setup() {
  beginLvglHelper(amoled);

  ui_init(); // Start lvgl ui.c

 lv_label_set_text(ui_TestTextLabel, "Here is my text");
}

void loop() {
    lv_task_handler();
    delay(5);
}`

Using the LVGL examples in the repository and on LVGL's website I learned which functions change whatever.  Like if you create a text label in SquareLine named TestTextLabel its object name will be ui_TestTextLabel.  And you can use lv_label_set_text(ui_TestTextLabel, "Here is my text"); to change the text.

Hope this helps.

OhioJoe, 2024-06-18:
I'm a 75 year old beginner. Is there a way to use a Slider to adjust the screen brightness on the Lilygo T4 S3 display? I have a Slider that responds to movement. I set the event -> Clicked to Call Function setBrightness. 

In the ui_events.c file: 
 `
void setBrightness(lv_event_t * e)
{
	// Your code here
}`

This is a function to adjust the brightness based on the Slider level from an Ardunio code.

`LilyGo_Class amoled;

static lv_obj_t *slider_label;
uint8_t level = 100;

static void slider_event_cb(lv_event_t *e)
{
  lv_obj_t *slider = (lv_obj_t *)lv_event_get_target(e);
  level = (uint8_t)lv_slider_get_value(slider);
  lv_label_set_text_fmt(slider_label, "%u", level);
  Serial.printf("Setting brightness to %i\r\n", level);
  amoled.setBrightness(level);
  lv_obj_align_to(slider_label, slider, LV_ALIGN_OUT_BOTTOM_MID, 0, 10);
}`

I would like to modify this to adjust screen brightness.  I know I will have to change the 'slider' name and comment out 
`  lv_obj_align_to(slider_label, slider, LV_ALIGN_OUT_BOTTOM_MID, 0, 10);`
As it is displayed on the screen at a label.

Hopefully some kind soul will be able to give me a simple explainations.

vpsmedeiros, 2024-06-18:
This is how my slider is setup to adjust the screen brightness on my 1.91" AMOLED touch.

In Squareline Studio my slider event trigger is set to VALUE_CHANGED.  Under the event trigger It calls a function 'BrightnessSliderEvent'.

my ui_events.cpp:
`void BrightnessSliderEvent(lv_event_t * e)
{
    uint8_t level = (uint8_t)lv_slider_get_value(ui_AdjustBrightSlider);
    amoled.setBrightness(level);
    lastBri = level;
}`

my main.cpp which I believe is pretty much a near copy of the factory example code:
`void setup() {
  LilyGo_Class amoled;
  beginLvglHelper(amoled);
  amoled.setBrightness(175);  // 0 to 255 is the range.
  int lastBri = amoled.getBrightness();

   // If you want the home button to turn the screen brightness all the way to 0
       amoled.setHomeButtonCallback([](void *ptr) {
        //Serial.println("\nHome key pressed");
        static uint32_t checkMs = 0;
        if (millis() > checkMs) {
            if (amoled.getBrightness()) {
                lastBri = amoled.getBrightness();
                amoled.setBrightness(0);
                //delay(500);
            } else {
                amoled.setBrightness(lastBri);
                //delay(500);
            }
        }
        checkMs = millis() + 500;`
};

OhioJoe, 2024-06-18:
Thanks for your suggestions.

I'm afraid my code is different on the LilyGO T4 S3 board. There is currently no driver in the TFT_eSPI code for this board. I am using SL version 1.4.1, setting for Arduino TFT_eSPI and Ardunio IDE 2.3.2
 The workaround is this in the ino file:
`#include <Arduino.h>
#include <LilyGo_AMOLED.h>
#include <LV_Helper.h>
#include "ui.h" //path to squareline studio export
#include "lvgl.h" //path to squareline studio export

LilyGo_Class amoled;

void setup() {
Serial.begin(115200); delay(2000);
  Serial.println("Running from:"); Serial.println(__FILE__);

  bool rslt = false;

  // Begin LilyGo  1.47 Inch AMOLED board class
  //rslt = amoled.beginAMOLED_147();

  // Begin LilyGo  1.91 Inch AMOLED board class
  //rslt =  amoled.beginAMOLED_191();

  // Begin LilyGo  2.41 Inch AMOLED board class
  //rslt =  amoled.beginAMOLED_241();

  // Automatically determine the access device
  rslt = amoled.begin();

  if (!rslt) {
    while (1) {
      Serial.println("The board model cannot be detected, please raise the Core Debug Level to an error");
      delay(1000);
    }
  }

  beginLvglHelper(amoled);
  amoled.setBrightness(175); // 0 to 255 is the range.

ui_init(); // Start lvgl ui.c

}

void loop() {
lv_task_handler();
delay(5);
}
`

When I added your code to the ui_events.c (I don't have a .CPP file), I get this error.

> Test/ui_events.c:11:70: error: 'amoled' undeclared (first use in this function); did you mean 'atoll'?
>    uint8_t level = (uint8_t)lv_slider_get_value(ui_BrightnessSlider); amoled.setBrightness(level);

[image]

The-Young-Maker, 2024-06-18:
From the little experience I have with LVGL, you should put it in your .ino file as the libraries for the display are defined in the .ino file. So you would just need to put this in your .ino file : 

```
static lv_obj_t *slider_label;
uint8_t level = 100;
static void slider_event_cb(lv_event_t *e)
{
lv_obj_t *slider = (lv_obj_t *)lv_event_get_target(e);
level = (uint8_t)lv_slider_get_value(slider);
lv_label_set_text_fmt(slider_label, "%u", level);
Serial.printf("Setting brightness to %i\r\n", level);
amoled.setBrightness(level);
lv_obj_align_to(slider_label, slider, LV_ALIGN_OUT_BOTTOM_MID, 0, 10);
}
lv_obj_add_event_cb(ui_BrightnessSlider, slider_event_cb, LV_EVENT_CLICKED, NULL);

```

Let me know if it works.

OhioJoe, 2024-06-18:
[The-Young-Maker](https://github.com/The-Young-Maker)

Thanks SO Much.  It works as it should now! I had to delete the following lines as they caused a reboot when the slider was moved.

` lv_label_set_text_fmt(slider_label, "%u", level);
 lv_obj_align_to(slider_label, slider, LV_ALIGN_OUT_BOTTOM_MID, 0, 10);`

Those lines are not needed now!

Again, Thanks So Much!

The-Young-Maker, 2024-06-18:
@OhioJoe no worries, glad it worked.

closed 2024-06-18 by The-Young-Maker