Xinyuan-LilyGO/LilyGo-AMOLED-Series #45: Can't use Lilygo T4-S3

julienCARR0, 2024-07-09:
I just bought a LilyGO T4-S3 and I've followed all the steps described to make it work. I tried on two computers and reinstalled Arduino IDE, but it still doesn't work. Once I complete the installation, I run the examples (lvgl get started, adjust brightness, etc.), but the screen is black. I've also tried to run the code "T4AMOLEDui-main" and it doesn't work. The screen is still black and it says ESP32TIME.h doesn't exist. So I tried to do a factory reset by installing the firmware, and then the factory program works fine. Can someone help me?

Thanks

lewisxhe, 2024-07-10:
ESP32TIME.h ? This library does not depend on this library file. Which example did you write? Adjust Core Debug Level to Debug and provide me with the log output.

julienCARR0, 2024-07-10:
I ran the TFT_eSPI_Sprite_graphicstest_small.ino example. it says "The current arduino version of TFT_eSPI does not support arduino 3.0, please change the version to below 3.0". how can i do that?

lewisxhe, 2024-07-11:
Please refer to step 4 in README. This library has requirements for esp32 version.

julienCARR0, 2024-07-11:
Yeah my bad it's working now. i feel a little stupid now ! Anyway thank you for your help.

closed 2024-07-11 by julienCARR0