Xinyuan-LilyGO/LilyGo-AMOLED-Series #48: Issue with T-display amoled V2. Not black screen or nor full power off.

fermintm, 2024-07-21:
Apparently these modules do not turn off the screen completely or leave it black. In the dark you can see that they are lit with a red tone and therefore consume more battery. I have 2 modules and both have the same error.
I have tried with several libraries and examples and the same error occurs.
I don't know if it is from the LilyGo_AMOLED library or it is a general problem.

lewisxhe, 2024-07-23:
AMOLED normally has no light after adjusting the brightness to 0. Try adjusting the brightness to 0 directly.
```
amoled.setBrightness(0)
```

fermintm, 2024-07-25:
> AMOLED normally has no light after adjusting the brightness to 0. Try adjusting the brightness to 0 directly.
> 
> ```
> amoled.setBrightness(0)
> ```

View link of my video. The issue is with latest Lilygo amoled Library.

closed 2024-09-08 by github-actions[bot]