Xinyuan-LilyGO/LilyGo-AMOLED-Series #52: T4-S3-2.41 - blinking red LED

regha, 2024-08-11:
Hi,
After uploading a program (GFX_HelloWorld) with Arduino IDE, nothing appears on th screen and red led is blinking (about 5 times a second)...
What can I do ?...

lewisxhe, 2024-08-14:
What about other examples? Testing factory firmware?

OhioJoe, 2024-08-14:
Did you press the Reset (rst) button after uploading?

carbuthn, 2024-09-09:
The T4 display is very nice but the flashing red led makes it un-useable for me. I wanted to buy a lot more but the led prevents that.

lewisxhe, 2024-09-10:
LED can be turned off [PPM_Example](https://github.com/Xinyuan-LilyGO/LilyGo-AMOLED-Series/blob/f88802d283480d835f9ef3ac9495d7854a575b27/examples/PPM_Example/PPM_Example.ino#L70)

carbuthn, 2024-09-17:
Is there a way to turn it off in esphome?

closed 2024-11-01 by github-actions[bot]

bernie26262, 2025-02-26:
So helpful. Thanks a lot for this info.