Xinyuan-LilyGO/LilyGo-AMOLED-Series #53: How do i setup the TFT.eSPI User_Seyup.h and User_Setup_Select.h i dont see the RM67162 Driver IC for the display

BluumDK, 2024-08-20:


lewisxhe, 2024-08-22:
I think it has been explained in the example that it can only be used as a frame buffer https://github.com/Xinyuan-LilyGO/LilyGo-AMOLED-Series/blob/f88802d283480d835f9ef3ac9495d7854a575b27/examples/TFT_eSPI_Sprite/TFT_eSPI_Sprite.ino#L34

BluumDK, 2024-08-22:
> I think it has been explained in the example that it can only be used as a frame buffer
> 
> https://github.com/Xinyuan-LilyGO/LilyGo-AMOLED-Series/blob/f88802d283480d835f9ef3ac9495d7854a575b27/examples/TFT_eSPI_Sprite/TFT_eSPI_Sprite.ino#L34

Ca i solely use lvgl or do i need tft.espi?

BluumDK, 2024-08-22:
> > I think it has been explained in the example that it can only be used as a frame buffer
> > https://github.com/Xinyuan-LilyGO/LilyGo-AMOLED-Series/blob/f88802d283480d835f9ef3ac9495d7854a575b27/examples/TFT_eSPI_Sprite/TFT_eSPI_Sprite.ino#L34
> 
> Ca i solely use lvgl or do i need tft.espi?

As well as this line:

#include <LilyGo_AMOLED.h> 

Is it just needed for the examples or is do i need to copy the files from the github and transfer them to my own projekt

Im very new to embedded im just trying to get the screen to say Hello World!

closed 2024-10-06 by github-actions[bot]