Xinyuan-LilyGO/LilyGo-AMOLED-Series #54: Amoled touch s3 series with black screen

Mathews-mart, 2024-08-21:
please i need help, I recently purchased a Lilygo Amoled Series S3 touch, it worked normally, but as soon as I used it for the first time (I uploaded any program, replacing the factory program) the Amoled screen stopped working, and no code that I put in the esp32 is solving it, I don't know how much you will charge for this help, but I'm really desperate because I'm Brazilian and the dollar here is very high and I need it urgently for a course completion project. I've tried everything and nothing worked

lewisxhe, 2024-08-22:
Did I follow the README instructions? See the FAQ for troubleshooting, troubleshooting the hardware first, then the software.

Mathews-mart, 2024-08-23:
yes I tried to follow the instructions in readme.md, but I got no results

lewisxhe, 2024-08-26:
I mean the binary file [here](https://github.com/Xinyuan-LilyGO/LilyGo-AMOLED-Series/blob/master/firmware%2FREADME.MD). Can it be displayed by writing it directly to the screen? If not, it can be judged as a hardware problem, and there is no need to find the cause from the software. If it is determined to be a hardware problem, please take a photo of the process of burning this file and the final result and send it to the dealer. If it is a hardware problem, they will replace it for you.

Mathews-mart, 2024-09-05:
Hi, I'm sorry for the delay, I tried this process but I didn't get any results.

Mathews-mart, 2024-09-05:
In fact, I bring good news, when I saved the finware again it worked, the problem is when I write another code on it, then it stops working, I saved the firmware on core 0 and the program on core 1

lewisxhe, 2024-09-06:
> In fact, I bring good news, when I saved the finware again it worked, the problem is when I write another code on it, then it stops working, I saved the firmware on core 0 and the program on core 1
** I saved the firmware on core 0 and the program on core 1**
What does this mean?

Mathews-mart, 2024-09-07:
It means that anything I record other than the firmware returns this error, and it's not even a simple hello world code that works, the screen goes back to black forever

lewisxhe, 2024-09-09:
1. Are you using the AMOLED-1.91-inch with touch?
2. Do you mean the firmware.bin? Rather than the firmware compiled by yourself through Arduino?

If everything works fine after you write firmware.bin, then it can be determined that the screen does not display due to the Arduino compilation environment or other problems. Please downgrade esp core to version 2.0.14, and then re-copy all the folders in the lib directory to arduino/libraries. For details, please see README, refer to README for settings, and then open Factory.ino and compile and upload again.

closed 2024-10-23 by github-actions[bot]