Xinyuan-LilyGO/LilyGo-AMOLED-Series #55: Does RM690B0 behaviour is the same than RM67162

dobodu, 2024-09-02:
Firstly I'd like to than you for this repo that help us so much is developping LVGL plateformio apps.
This issue question is clearly not an issue to your repository but more a question of understanding.

I've been developping a [micropthon library](https://github.com/dobodu/RM690B0_Micropython_QSPI) holding the RM690B0 display driver for a T4-S3 device.
My repository was inspired by another dev who build the initial release holding the RM67162 driver of T-display S3 devices.
We progressively implemented some function inspired by Rush Hughes ST7789 library, aiming to create the lightest library allowing
to program micropython functionality to those AMOLED wonderfull devices and make something easily portable between PICO and ESP32, handing ST7789 to RM67162 and RM690B0. That would be great isn't it ? However....

Except for initialisation purpose, the datasheet of both display driver look the same. However, while trying to use the same functions (a basic font printing helloworld example), I realized that I faced a kind of 45° switched font when using RM690B0 library everytime the X or Y coordonnates doesn't fullfill the CASET or RASET requirements (must be multiples of 2). I couldn't get the same issue while using the RM67162 library which used the exact same memory copy functions. Strange isn't it ? So here is my question :

Did you faced this kind of issue while developping your library (I know is uses LVGL functions that might benefit some code to handle this kind of issue). What's your opinion on how to handle this issue ? 

Of course, I feel like spamming you repository, feel free to remove this issue if needed. But I would really appreciate a small help.

lewisxhe, 2024-09-05:
The problem you encountered is mentioned here
https://github.com/Xinyuan-LilyGO/LilyGo-AMOLED-Series/issues/16

dobodu, 2024-09-05:
> 
> The problem you encountered is mentioned here
> https://github.com/Xinyuan-LilyGO/LilyGo-AMOLED-Series/issues/16
> 

Oh thanks, it confirms the diagnostic I made.

I've been adding yesterday options to deal with odd/even values (in fact the same solution that the one in the thread you mention). It involves some minimal border effect... 

I definitely do not want to use full refresh all the time for performance purpose.

closed 2024-10-20 by github-actions[bot]