Xinyuan-LilyGO/LilyGo-AMOLED-Series #58: ESP32 *below* V3.0 ?

byte-for-byte, 2024-09-27:
Hello,
Thanks for the great work so far!!

I am curious:
a) why is there the <V3.0 restriction?
b) is there a plan to remove the <V3.0 restriction?

> Install [Arduino ESP32 V 2.0.5 or above and below V3.0]

Thanks in advance

lewisxhe, 2024-09-27:
Because 3.0 was in the dev state when this library was written, the API has changed too much and is incompatible. In addition, the TFT_eSPI example is currently only applicable to version 2.0.14, and will be updated to support 3.0 and above later.

byte-for-byte, 2024-09-28:
Thanks for the details and the quick answer! :)

I assume there is some kind of timeline or milestone for that at LilyGo? I am wondering if I should wait for your updates or look into another display that works with the up-to-date esp32 core installation?

thanks in advance

lewisxhe, 2024-10-01:
I can't give you exact times because I maintain many repositories at the same time.

lewisxhe, 2024-10-04:
I spent some time to test whether the factory can run normally on version v3.0.5. After testing, it can run on versions above V3.0 with simple modifications. At present, I have only tested the factory example. Other examples have not been tested. You can test them when you have time.

lewisxhe, 2024-11-05:
Currently only LVGL V8 is available
[image]

byte-for-byte, 2024-11-05:
Hi, thanks for the quick reply.
Tbg, I am really disappointed with the LilyGo software support, but maybe I am expecting too much.

From the changelog of lvgl v9.0 (released Jan 22 !),  at https://docs.lvgl.io/9.0/CHANGELOG.html, I see that there were breaking changes in the API, e.g.: "lv_msg is removed and replaced by lv_observer".

Do you see a chance that the the Liligo examples are migrated to the latest lvgl API in the near future and all examples would compile with the latest esp32 arduino core packages setup?

thanks,
best,

lewisxhe, 2024-11-06:
There will be no major updates for the time being. I will not only maintain this repository, but I will also update lvgl when I have time later.

closed 2024-12-22 by github-actions[bot]