Xinyuan-LilyGO/LilyGo-AMOLED-Series #59: Charging problem

scuba-hacker, 2024-09-28:
I am using a lithium cell to power the T4 Amoled 2.4 inch device through the battery socket on the back of the board. Battery power is delivered to the board without problems, the plug/socket and wiring are secure and moving the wires or touching with plastic tweezers the battery socket/plug doesn't interrupt power. The Red LED flashes as expected to show not charging.

I then connect a 5V USB supply through the header pins to charge the battery. When I first started using the screen earlier in the year the red LED would go steady and stay on until charging of the battery was finished whereupon the LED flashed once again - so far so good.

Now something strange is happening with the socket on the board. When I charge the battery it requires manipulation of the plug/socket/wiring to make the LED go steady red to make the battery charge and it can take a lot of fiddling to get the charging active again. The slightest touch makes the LED flash and my USB power meter shows that the battery has stopped charging. I then have to fiddle again with the socket and then the light, if I'm lucky, will go steady and charging recommences.

I know that the wiring/socket/plug is good for supplying power from the battery to the board, but there is a definitely a problem when the battery is charging.

When not charging current flow is 170mA.
When charging current flow is 410mA.

lewisxhe, 2024-10-01:
From your description, it is possible that the switch or the cable caused the charging problem, right? You said the switch was broken, so the switch may be more of a problem

scuba-hacker, 2024-10-01:
> From your description, it is possible that the switch or the cable caused the charging problem, right? You said the switch was broken, so the switch may be more of a problem

there’s no problem with a cut out when running on battery power alone. I never touch the remnants of the switch, it stays permanently in the on position and i’ve never had a spurious loss of power on battery. the board is mounted in an enclosure and is charged with the enclosure open so the board is secure. My suspicion is the battery socket on the board may have two pcb layers connected to it and one is for charging and this has some sort of bad connection?

I haven’t tried connecting the battery through the header pins yet as that needs cutting the battery plug off as i don’t have another a jst ph male socket. I assume the battery can be charges through the header?

lewisxhe, 2024-10-04:
You can try connecting the battery here to see if the internal connection of the PCB is unreliable. This problem can be ruled out because it is unlikely
[image]

scuba-hacker, 2024-10-22:
I got hold of a new battery connector which fits more snugly to the board socket compared to the original and it's fixed the charging problem. Very strange that only charging was affected. Issue can be closed, thanks for the advice.

closed 2024-10-23 by lewisxhe