Xinyuan-LilyGO/LilyGo-AMOLED-Series #61: GPIO18 is not free to use!

Katana1234, 2024-10-04:
Hi,

looking at the schematic GPIO18 looks like to be a pin that can be used freely:
[image]

When in reality it is connected to the display.
[image]

Why can't this be seen in the schematic???
I fried one of my displays because i thought it was an unused pin!

lewisxhe, 2024-10-04:
It was added to TEPin later, but the schematic was not updated. Is your screen damaged?

Katana1234, 2024-10-04:
> It was added to TEPin later, but the schematic was not updated. Is your screen damaged?

yes it is. i was using it for an analog reading, but due to the fact that it was connected to the display it drew a very high current until the board or the screen broke...

lewisxhe, 2024-10-04:
Please contact customer service and I will arrange to send you another one.

lewisxhe, 2024-10-05:
Schematic updated, issue closed

closed 2024-10-05 by lewisxhe