Xinyuan-LilyGO/LilyGo-AMOLED-Series #62: T4-S3 (H678) SD-card pins reuse?

pike2k, 2024-10-09:
There's not many free GPIO's on this board, so I wonder if GPIO 01, 02, 03, 04 which are all connected to the onboard SD-card reader can be reused?

It seems like this filevsd_diskio.cpp tries to init the SD-card. I don't want this SD functionality, can it be disabled somewhere?

lewisxhe, 2024-10-11:
After amoled.begin(), call SPI.end() to end SPI binding

pike2k, 2024-10-11:
Thanks Lewis, I tried adding SPI.end(); after amoled.begin(); in setup() . 
However, It did not stop the SD init:

[  4641][W][sd_diskio.cpp:174] sdCommand(): no token received
[  4744][W][sd_diskio.cpp:174] sdCommand(): no token received
[  4844][W][sd_diskio.cpp:174] sdCommand(): no token received
[  4944][E][sd_diskio.cpp:199] sdCommand(): Card Failed! cmd: 0x00
[  4944][W][sd_diskio.cpp:516] ff_sd_initialize(): GO_IDLE_STATE failed
[  4945][E][sd_diskio.cpp:802] sdcard_mount(): f_mount failed: (3) The physical drive cannot work
[  4954][W][sd_diskio.cpp:174] sdCommand(): no token received
[  5059][W][sd_diskio.cpp:174] sdCommand(): no token received
[  5160][W][sd_diskio.cpp:174] sdCommand(): no token received
[  5261][E][sd_diskio.cpp:199] sdCommand(): Card Failed! cmd: 0x00
[  5261][E][LilyGo_AMOLED.cpp:484] beginAMOLED_241(): Failed to dected SDCard!

lewisxhe, 2024-10-12:
I updated the initialization method of T4-S3. Please use beginAMOLED_241(false) to initialize instead of begin(). This way you can freely control whether sd should be initialized.

```c
bool disable_sd = true;
bool disable_charge_led = true;
amoled.beginAMOLED_241(disable_sd,disable_charge_led);
```

pike2k, 2024-10-12:
Thanks Lewis. I will test it as soon as I figure out how I can use the master of Lilygo-AMOLED-Series in PlatformIO instead of 1.08 that I was using previously!

pike2k, 2024-10-13:
Lewis, I have not been able to figure out how to use Master version of this library with PlatformIO. 
I believe it goes against basic principles how you are supposed to use PlatformIO?
Is there any chance that you could release 1.0.9?

lewisxhe, 2024-10-14:
Did you not figure out how to use platfromio? Did you read the README?

1.0.9 will not be released, I will release 1.1.0. I will stop for a few days to see if there are other problems. If there are no other problems, I will push V1.1.0 to arduino library today.

pike2k, 2024-10-17:
I did a quick test with one of the involved SD IO's and it was free, so closing unless something pops up :)

closed 2024-10-17 by pike2k