Xinyuan-LilyGO/LilyGo-AMOLED-Series #63: Pinout clarification

Trolomite, 2024-10-12:
Hi, I'm new with ESP 32 and I'm trying to figure out how to communicate with eternal devices (wired), via I2C or SPI. Ideally SPI.
I have the LilyGo T4 S3 display. I'm assuming that there's no SPI available because of the display & SD card slot. Is that correct?
If external SPI is not available, how about I2c? which pins do I use for that?
It's not very clear from the documents (as a beginner).
Any help would be much appreciated. 

Thanks

pike2k, 2024-10-12:
For I2C you have Qwiic, it's a convenient standard. 
I have developed a breadboard adapter for T4-S3 that I needed for my project.

[image]

lewisxhe, 2024-10-14:
As long as the GPIO port is not used, it can be freely assigned to SPI.
Just specify the GPIO Pin during initialization, like this
https://github.com/Xinyuan-LilyGO/LilyGo-AMOLED-Series/blob/992ac04edd961a064e8b2f0e512a3810ba33f687/src/LilyGo_AMOLED.cpp#L617

I2C is the same, but you can't conflict with the Wire used in the library, you need to use Wire1.begin(sda,scl) . See the Qwiic example for details

closed 2024-11-27 by github-actions[bot]