Xinyuan-LilyGO/LilyGo-AMOLED-Series #64: LilyGO T3 S4 any of the examples are compiled with: Compilation error Also Factory

Robiv8, 2024-10-12:
New LilyGO T3 S4 with Arduino IDE, all Librarys loaded and any of the examples are not compiled.
By the way: Blink Sample compile OK ;-)

lewisxhe, 2024-10-14:
You said it can't compile? Did you follow the instructions in the README?

Robiv8, 2024-10-15:
Yes, all in Instructionsare are followed.
There is a code of this Member on GitHub and this works without any problems.
https://github.com/VolosR/T4AMOLEDui

None of the LilyGo-Amoled-Series examples can be installed. All report any errors.

Sample = Factory.ino

`/private/var/folders/1p/gwd6257d4091gd9p35k7srmr0000gn/T/.arduinoIDE-unsaved2024915-627-tisnf6.oa89/Factory/Factory.ino: In function 'void WiFiEvent(arduino_event_id_t)':
/private/var/folders/1p/gwd6257d4091gd9p35k7srmr0000gn/T/.arduinoIDE-unsaved2024915-627-tisnf6.oa89/Factory/Factory.ino:367:9: error: 'lv_msg_send' was not declared in this scope
         lv_msg_send(WIFI_MSG_ID, NULL);
         ^~~~~~~~~~~
/private/var/folders/1p/gwd6257d4091gd9p35k7srmr0000gn/T/.arduinoIDE-unsaved2024915-627-tisnf6.oa89/Factory/Factory.ino:367:9: note: suggested alternative: 'lv_fs_seek'
         lv_msg_send(WIFI_MSG_ID, NULL);
         ^~~~~~~~~~~
         lv_fs_seek
/private/var/folders/1p/gwd6257d4091gd9p35k7srmr0000gn/T/.arduinoIDE-unsaved2024915-627-tisnf6.oa89/Factory/Factory.ino: In function 'void updateCoin360Task(void*)':
/private/var/folders/1p/gwd6257d4091gd9p35k7srmr0000gn/T/.arduinoIDE-unsaved2024915-627-tisnf6.oa89/Factory/Factory.ino:557:37: error: 'lv_msg_send' was not declared in this scope
                                     lv_msg_send(COIN_MSG_ID, &coinData[i]);
                                     ^~~~~~~~~~~
/private/var/folders/1p/gwd6257d4091gd9p35k7srmr0000gn/T/.arduinoIDE-unsaved2024915-627-tisnf6.oa89/Factory/Factory.ino:557:37: note: suggested alternative: 'lv_fs_seek'
                                     lv_msg_send(COIN_MSG_ID, &coinData[i]);
                                     ^~~~~~~~~~~
                                     lv_fs_seek
/private/var/folders/1p/gwd6257d4091gd9p35k7srmr0000gn/T/.arduinoIDE-unsaved2024915-627-tisnf6.oa89/Factory/Factory.ino: In function 'void updateWeatherTask(void*)':
/private/var/folders/1p/gwd6257d4091gd9p35k7srmr0000gn/T/.arduinoIDE-unsaved2024915-627-tisnf6.oa89/Factory/Factory.ino:728:17: error: 'lv_msg_send' was not declared in this scope
                 lv_msg_send(WEATHER_MSG_ID, &weatherApi);
                 ^~~~~~~~~~~
/private/var/folders/1p/gwd6257d4091gd9p35k7srmr0000gn/T/.arduinoIDE-unsaved2024915-627-tisnf6.oa89/Factory/Factory.ino:728:17: note: suggested alternative: 'lv_fs_seek'
                 lv_msg_send(WEATHER_MSG_ID, &weatherApi);
                 ^~~~~~~~~~~
                 lv_fs_seek
/private/var/folders/1p/gwd6257d4091gd9p35k7srmr0000gn/T/.arduinoIDE-unsaved2024915-627-tisnf6.oa89/Factory/gui.cpp: In function 'void update_date(lv_timer_t*)':
/private/var/folders/1p/gwd6257d4091gd9p35k7srmr0000gn/T/.arduinoIDE-unsaved2024915-627-tisnf6.oa89/Factory/gui.cpp:80:9: error: 'lv_msg_send' was not declared in this scope
         lv_msg_send(TEMPERATURE_MSG_ID, NULL);
         ^~~~~~~~~~~
/private/var/folders/1p/gwd6257d4091gd9p35k7srmr0000gn/T/.arduinoIDE-unsaved2024915-627-tisnf6.oa89/Factory/gui.cpp:80:9: note: suggested alternative: 'lv_fs_seek'
         lv_msg_send(TEMPERATURE_MSG_ID, NULL);
         ^~~~~~~~~~~
         lv_fs_seek
/private/var/folders/1p/gwd6257d4091gd9p35k7srmr0000gn/T/.arduinoIDE-unsaved2024915-627-tisnf6.oa89/Factory/gui.cpp: In lambda function:
/private/var/folders/1p/gwd6257d4091gd9p35k7srmr0000gn/T/.arduinoIDE-unsaved2024915-627-tisnf6.oa89/Factory/gui.cpp:131:80: error: invalid conversion from 'void*' to 'lv_obj_t*' [-fpermissive]
             lv_obj_t *bg = (lv_obj_t *)lv_obj_get_user_data(lv_event_get_target(e));
                                                             ~~~~~~~~~~~~~~~~~~~^~~
In file included from /Users/robi/Documents/Arduino/libraries/lvgl/src/../lvgl.h:39,
                 from /Users/robi/Documents/Arduino/libraries/lvgl/src/lvgl.h:16,
                 from /private/var/folders/1p/gwd6257d4091gd9p35k7srmr0000gn/T/.arduinoIDE-unsaved2024915-627-tisnf6.oa89/Factory/gui.h:12,
                 from /private/var/folders/1p/gwd6257d4091gd9p35k7srmr0000gn/T/.arduinoIDE-unsaved2024915-627-tisnf6.oa89/Factory/gui.cpp:11:
/Users/robi/Documents/Arduino/libraries/lvgl/src/../src/core/lv_obj.h:335:40: note:   initializing argument 1 of 'void* lv_obj_get_user_data(lv_obj_t*)'
 void * lv_obj_get_user_data(lv_obj_t * obj);
                             ~~~~~~~~~~~^~~
/private/var/folders/1p/gwd6257d4091gd9p35k7srmr0000gn/T/.arduinoIDE-unsaved2024915-627-tisnf6.oa89/Factory/gui.cpp: In function 'void createTimeUI(lv_obj_t*)':
/private/var/folders/1p/gwd6257d4091gd9p35k7srmr0000gn/T/.arduinoIDE-unsaved2024915-627-tisnf6.oa89/Factory/gui.cpp:152:5: error: 'lv_label_set_recolor' was not declared in this scope
     lv_label_set_recolor(time_label, 1);
     ^~~~~~~~~~~~~~~~~~~~
/private/var/folders/1p/gwd6257d4091gd9p35k7srmr0000gn/T/.arduinoIDE-unsaved2024915-627-tisnf6.oa89/Factory/gui.cpp:152:5: note: suggested alternative: 'lv_led_set_color'
     lv_label_set_recolor(time_label, 1);
     ^~~~~~~~~~~~~~~~~~~~
     lv_led_set_color
/private/var/folders/1p/gwd6257d4091gd9p35k7srmr0000gn/T/.arduinoIDE-unsaved2024915-627-tisnf6.oa89/Factory/gui.cpp:223:9: error: 'lv_msg_subsribe_obj' was not declared in this scope
         lv_msg_subsribe_obj(COIN_MSG_ID, label, NULL);
         ^~~~~~~~~~~~~~~~~~~
/private/var/folders/1p/gwd6257d4091gd9p35k7srmr0000gn/T/.arduinoIDE-unsaved2024915-627-tisnf6.oa89/Factory/gui.cpp: In lambda function:
[... cut ...]
                                                LV_SUBJECT_TYPE_COLOR
Mehrere Bibliotheken wurden für "LilyGo_AMOLED.h" gefunden
  Benutzt: /Users/robi/Documents/Arduino/libraries/LilyGo-AMOLED-Series
  Nicht benutzt: /Users/robi/Documents/Arduino/libraries/coolUI
Mehrere Bibliotheken wurden für "SD.h" gefunden
  Benutzt: /Users/robi/Library/Arduino15/packages/esp32/hardware/esp32/2.0.17/libraries/SD
  Nicht benutzt: /Users/robi/Library/Arduino15/libraries/SD
exit status 1

Compilation error: 'lv_msg_send' was not declared in this scope`

Robiv8, 2024-10-15:
So!
After a lot of googling I found out that it is due to the LVGL version.
If you downgrade it to 8.4.0 then it works so far, I haven't tested everything yet.
It's sad that such a great display is released without concrete instructions and information. You have to search for everything on the internet.

closed 2024-10-15 by Robiv8

lewisxhe, 2024-10-16:
You may have overlooked that all the relevant instructions are written in the README
[image]
[image]