Xinyuan-LilyGO/LilyGo-AMOLED-Series #66: Location of JST-GH 1.25mm & T4-S3-Sheild H731 Battery solder pad connected to header pin 3V3 or 5V?

hohaha045, 2024-10-31:
@lewisxhe @LilyGO 

Would like to ask about the location of the batter connector JST-GH 1.25mm, especially how many mm between the edge of the board? I've checked the dxf, and stp files, but the JST-GH1.25 is not on those files. 

And for the Sheild H731, are the battery solder pad connected to the header `BAT`? or somewhere else? in the official image, I can t see any `BAT+` or `BAT-` but only two `BAT` header pins, and also, would the shield battery solder pad work with the on/off switch? or only possible to use it with the onboard JST-GH connector?

lewisxhe, 2024-11-01:
Already added

lewisxhe, 2024-11-01:
[image]
The pad is directly connected to the power chip and cannot be connected through a switch

closed 2024-12-16 by github-actions[bot]