Xinyuan-LilyGO/LilyGo-AMOLED-Series #70: Amoled touch version - turn off display and and can keep the device always on?

zorro1011, 2024-11-21:
hi
I received the product and I would like to know if there is a way to turn the display off and on again.
for now I'm doing it like this but I'm not sure if it's the right way

`    amoled.setHomeButtonCallback([](void *ptr) {
        static uint32_t checkMs = 0;
        if (millis() > checkMs) {
            if (displayOn) {
                amoled.setBrightness(0); // Spegne il display
                Serial.println("Display spento");
            } else {
                amoled.setBrightness(128); // Riaccende il display con luminosità media
                Serial.println("Display acceso");
            }
            displayOn = !displayOn; // Alterna lo stato del display
            checkMs = millis() + 200;
        }
    }, NULL);`

I would also like to use the T-display as an anti-theft control unit and I need to leave the device always on with espnow working.
I noticed that, with the display "off" it consumes 90mA and heats up a little (not much).
I can use it on 24 hours a day, 365 days a year

tanks
best regards

zorro1011, 2024-11-21:
sorry :-(

1.91" T-Display S3 Amoled Touch

zorro1011, 2024-11-22:
first of all thanks for answering me

yes i know. i used it in the code above.
the thing i dont know is if there is a specific command to turn off and on the display (as you can see i used the brightness adjustment - but i dont think thats the right thing)

the second question was if i can leave the T-display always on (24 hours a day 365 days a year)? (i have to make an alarm control unit (so with espnow always on in reception waiting for any alarms from an esp32c3 of the windows)

thanks again
best regards

lewisxhe, 2024-11-22:
1. Although its name is to adjust the brightness, when the brightness is set to 0, the display is actually completely off, which is different from the TFT backlight screen, so you can rest assured.

2. If you want to display for a long time, I think it is best to refresh other interfaces after a period of time, such as half an hour or an hour, to avoid screen burn-in, although I haven't encountered it yet.

zorro1011, 2024-11-22:
1) Ok then it's fine like this.
2) The AMOLED display will be off the entire time (at least I hope so, as it will only turn on if there's a break-in alarm).

So, I should be all set like this.

thanks so much
best regards

zorro1011, 2024-11-23:
Hi again
I read the code of the LilyGo_AMOLED.h and .CPP library and I found the settings to actually turn off the display.
In fact, unlike brightness 0, where the consumption in mA remains identical to that with the display on with brightness 128, using this code the consumption decreases by 20mA.
this is the code that I modified:
```
    amoled.setHomeButtonCallback([](void *ptr) {
        static uint32_t checkMs = 0;
        if (millis() > checkMs) {
            if (displayOn) {
                amoled.sleep(); // Spegne il display usando la funzione sleep
                Serial.println("Display spento");
            } else {
                amoled.wakeup(); // Riaccende il display usando la funzione wakeup
                delay(100);
                Serial.println("Display acceso");
            }
            displayOn = !displayOn; // Alterna lo stato del display
            checkMs = millis() + 200;
        }
    }, NULL);
```
the problem is that it turns off perfectly but I can't turn it on anymore and the strange thing is that if I press the UI buttons (which I know where they are physically even if you can't see them) I get confirmation in the serial monitor that I pressed them.
Always on the serial monitor I read "Display off, Display on" at every press of the home button.
So it seems they all work but the display remains off.

It became a matter of stubbornness, so could someone help me solve it?

lewisxhe, 2024-11-24:
amoled.sleep(); sets all peripherals to sleep mode, not just turning off the screen, but turning off all.

lewisxhe, 2024-11-24:
I added disp_sleep and disp_wakup methods, which are to set the screen to sleep and wake up, I haven't tested this, you can try it

zorro1011, 2024-11-24:
works perfectly!!!
even the current consumption drops by 20mA
THANKS SO MUCH

 is there a way to disable the touch screen (only the display part) leaving the home button working

lewisxhe, 2024-11-24:
Please see the commit information, this should be able to achieve your function. Calling disableTouch will always return the number of presses as 0, and will not actually disable the touch function.
Calling enableTouch will normally enable touch

zorro1011, 2024-11-24:
it works
THANKS SO MUCH

best regards

vpsmedeiros, 2024-11-28:
This addition of disp_sleep/wake and touch does work for me.  Thank you for this.  I was looking for a way to completely shutoff the display because before it would still show a very dim light under night vision.

I noticed when disp_sleep is called my display is no longer refreshing.  For example, I have a function that is called every 20 seconds for checking the voltage and changing my battery icon and voltage readout "lv_label_set_text_fmt(ui_VoltageLabel, "%.2fv", (BattVolt * .001));".  The text and icon image does not change when the disp_sleep is active.  When I touch the home button to turn the display on, the battery reading will be as when I turned off the screen until the battery check loop is called again.

Is there a way to fix that or because the display is shutoff lvgl can no longer apply changes?

lewisxhe, 2024-11-28:
It is useless to refresh the screen after disp_sleep is set, disp_wakeup must be called before refreshing

RP6conrad, 2024-12-08:
To minimize deepsleep current, you have to activate the sleep mode from the displaydriver, and sleepmode from the touchdriver ! I use these commands : 
lcd_sleep(); (display driver, lib rm67162.h))
touch.enableAutoSleep(); (touch driver, lib TouchDrvCSTXXX.hpp) 
Current is now  180 µA in deepsleep.

closed 2025-01-21 by lewisxhe