Xinyuan-LilyGO/LilyGo-AMOLED-Series #71: Unable to Use T-Display-S3 as USB Host for Keyboard Input

devilsbane98, 2024-11-22:
I am trying to connect a keyboard to the LilyGO T-Display-S3 AMOLED as an input device using USB Host mode. However, I am encountering the following issues:

The USB DM and USB DP pins, which are mentioned in the schematic, are not exposed anywhere on the board. This makes it unclear how to connect an external USB device directly.
The onboard USB-C connector does not recognize the keyboard when connected, despite attempting to configure the board for USB Host mode.
I have tried using:

ESP-IDF's USB Host stack, and
The ESPUsbHost library in Arduino,
but neither approach has worked for detecting or initializing the keyboard.
It seems there might be limitations in using the onboard USB-C connector or configuring the board for USB Host mode. Is there any official documentation or workaround for enabling USB Host functionality on this board? Additionally, is there a hardware solution (e.g., exposing DM/DP pins) to connect USB devices directly?

Any help or guidance on this matter would be greatly appreciated!

lewisxhe, 2024-11-24:
USB-C cannot provide power to the outside world, so you cannot recognize the device using USB-C.
Have you tried to power the keyboard additionally?

devilsbane98, 2024-11-24:
My device is powered externally not via the USB port on the board

lewisxhe, 2024-11-25:
Can you detect the keyboard now?

devilsbane98, 2024-11-25:
The power is fine, but the board cannot connect to it. I have the power and GND pins of the keyboard connected to the same external power source as the board and the D+ and D- pins are connected via an adapter to the board's USB-C port.

lewisxhe, 2024-11-26:
You should use analogReadMilliVolts(35) * 2

lewisxhe, 2024-11-26:
USB-C is directly connected to D+, D-, without passing anything in the middle, so this problem may occur in some settings. I wrote an example for another board before, you can refer to it

https://github.com/Xinyuan-LilyGO/T-Display-S3-Pro/blob/master/examples/USB_HID_Example/USB_HID_Example.ino

lewisxhe, 2024-11-26:
I added examples that you can check out. I'm using T4 but it's the same as the others.

lewisxhe, 2024-11-27:
Did you follow the instructions in my video?

lewisxhe, 2024-11-28:
Are you monitoring the serial via the Qwiic 43 44 pins? The hardware is identical, so I suspect your keyboard doesn't support USB HID.

lewisxhe, 2024-11-28:
OK, you need to provide me with the PN number on your modem label sticker, and you need to update to the latest firmware, and then use the log capture tool to capture the modem log, and then submit it to simcom for analysis.

closed 2025-01-11 by github-actions[bot]