Xinyuan-LilyGO/LilyGo-AMOLED-Series #72: T4-S3 Display no powering display from battery

klone52, 2024-11-23:
Hi , I'm having an issue where using any sketch I see the display powered On and everuthing ok if I plug the USB before , and stay On when unplugged with the battery connected. But if I only use the battery, the pcb is  powered but the display dosn't turn on.

lewisxhe, 2024-11-24:
I need to check the PCB and then tell you the test method. I will reply you tomorrow.

lewisxhe, 2024-11-25:
Connect the battery and measure whether there is battery voltage at these two ends

[image]

klone52, 2024-11-25:
Hi ! with my battery on 3.8V i see 2.49V on this point

lewisxhe, 2024-11-26:
OK, I suspect that this MOS is damaged. Connect D and S according to the instructions in the figure below and see if the power supply is normal.
[image]

dobodu, 2024-12-23:
Hi lewis,

My T4-S3 has lost display that was working well a few days ago. I power-it up with USB.
Would you give me an advice if I need to check anything on the mainboard ?
The display itself seems to be working (without light) since when I draw lines, I have no error and the
correct time to proceed. But no pixel enlighten...

lewisxhe, 2024-12-24:
There is a high possibility that the screen is damaged, and the screen power supply voltage needs to be tested. This test point must be tested by removing the motherboard, and the screen must be connected during the test.
[image]

dobodu, 2024-12-24:
Thank you Lewis, I'll try the test point.
I hope T3 S4 with enclosure is not hard to dissassembly.

dobodu, 2024-12-26:
Hi Lewis,
I've been checkin the testpoint. A expected ELVSS = -3.6V and ELVDD = +3.6V.
Checking the schematics, I think EL6804 is powering correctly the OLED display which is dead i believe.
Am I right ?

lewisxhe, 2024-12-27:
ELVSS and ELVDD are in the normal voltage range. If the screen doesn't light up, the only possibility is that the screen is broken, or maybe the motherboard. Have you tested the factory firmware?

dobodu, 2024-12-27:
Yes, uploaded factory firmware, board is getting on load but no display...

closed 2025-02-10 by github-actions[bot]