Xinyuan-LilyGO/LilyGo-AMOLED-Series #73: T-display S3 Amoled touch 1.91" -  use Lipo to also power other modules

zorro1011, 2024-12-03:
Hi

I think I understood that I can use a lipo as a UPS in the case the 220V current of the usb-c power supply is missing.
if I'm wrong correct me.

I would like to understand if I can use this system in some way to also power the SIM800L EVB module.

thanks
best regards

lewisxhe, 2024-12-04:
The board cannot be used to directly power SIM800L. SIM800L requires at least 2A of power, and the board cannot provide such a large current. Only an external power supply can be used.

zorro1011, 2024-12-05:
thanks for the reply and sorry for my delay.
I measured my SIM800L EVB module and I don't detect a consumption of 2A (as per specifications) maybe because sending an SMS doesn't consume that much.
it's also true that I took the measurement with a normal multimeter which is not useful for measuring short peaks.

however I had already ordered a UPS 5V 3A from aliexpress, immaginando di non risolvere direttamente.
I will use this UPS to power both the SIM800L module and the T-Display

thanks again
best regards

zorro1011, 2024-12-07:
yes I read it, thanks

I wonder how the "LILYGO T-SIM7000G" can work with LiPo.
maybe the 7000G module consumes less

lewisxhe, 2024-12-09:
7000g can be powered directly by battery. SIM7000G consumes less current than SIM800L. Compared with SIM800L, SIM7000G is much more stable than SIM800L.

zorro1011, 2024-12-09:
I assumed so
thanks
best regards

closed 2025-01-21 by lewisxhe