Xinyuan-LilyGO/LilyGo-AMOLED-Series #74: Error with XPowersPPM

MichelReij, 2024-12-11:
When loading any of the tft_espi examples I get this error:

```
/Users/michelreij/Documents/Arduino/libraries/LilyGo-AMOLED-Series/src/LilyGo_AMOLED.h:305:22: error: expected class-name before ',' token
  305 |     public XPowersPPM,
      |       
```

Has anyone else expected this before? It also happens with some older code that I have not touched for half a year or so (which also uses tft_espi).

MichelReij, 2024-12-11:
More information:
I'm setting up a new Mac (ARM), environment is the Arduino IDE 2.3.4:
```
Version: 2.3.4
Date: 2024-12-03T09:58:49.900Z
CLI Version: 1.1.1
```
Lilygo-AMOLED-Series library is 1.1.0 (installed through Arduino IDE library manager), lvgl is at 8.4.0

Example to test with: TFT_eSPI_Sprite

Switching back to LilyGo_AMOLED-Series 1.0.8 solves the issue.

lewisxhe, 2024-12-12:
This is an update of the upstream dependency library. The error is caused by not updating in time. Downgrade first.
Library Search -> XPowersLib -> v0.2.6 -> Install

MichelReij, 2024-12-13:
Thank you @lewisxhe

closed 2024-12-13 by MichelReij