Xinyuan-LilyGO/LilyGo-AMOLED-Series #75: 'LilyGo_AMOLED::XPowersPPM' has not been declared

MikeyMoMo, 2024-12-20:
This is code that was loaded onto a T4S3 a month ago with no problem.  I needed to make a minor change and tried to recompile it today and got this odd message.  I backup up to AMOLED.h 1.0.8.  1.1.0 also gives this error as does 1.2.0.  I don't know where to go on this.  AMOLED.h loads XPowersLib.h.  I also tried to load it and no change on the error message.  How do I make this work once again?

```
#include<LilyGo_AMOLED.h>
LilyGo_Class amoled;

amoled.XPowersPPM::disableStatLed();

Utilities:414:12: error: 'LilyGo_AMOLED::XPowersPPM' has not been declared
     amoled.XPowersPPM::disableStatLed();  // Turn off the flashing charge LED.
            ^~~~~~~~~~
```
Used to work.  Turned the charge light off.  No compile error.  What happened???
Thanks for any pointers to make this compile again.
Mike

lewisxhe, 2024-12-20:
https://github.com/Xinyuan-LilyGO/LilyGo-AMOLED-Series/issues/74

lewisxhe, 2024-12-20:
Arduino IDE does not have version management. Please pay attention to the version requirements of the dependent libraries, which can be seen here.
https://github.com/Xinyuan-LilyGO/LilyGo-AMOLED-Series/blob/master/library.json

lewisxhe, 2024-12-21:
There is another problem. The PPM example you are running has changed. Please check the example

closed 2025-01-21 by lewisxhe