Xinyuan-LilyGO/LilyGo-AMOLED-Series #78: Issues with compiling any code for T4 S3. Issues after uploading.

whzbngry, 2024-12-27:
**Background:**
Hardware - T4 S3 (v1.1 board). 
Arduino IDE:
- Version: 2.3.2
- CLI Version: 0.35.3

**Versions:**
[image]
[image]
[image]
[image]
[image]
[image]

**Problem:**
- I have tried almost all examples from LilyGo Amoled Series -> lvgl, or Lvgl_images, or Ardunio GFX examples or, Tft_Espi examples.
- None of the examples compile for some reason.
- The "Arduino ESP32 Boards" and "eps32" version work with my other projects so I haven't updated them yet. Let me know if that is the cause of the problem

Complier shows the following error:
[image]

The only thing that works is:
- Flashing the [firmware](https://github.com/Xinyuan-LilyGO/LilyGo-AMOLED-Series/tree/master/firmware) file - this is the only thing that fully works. Even the Factory.ino code doesn't compile and shows the same error as shown above.
- Uploading a very basic code like (only uploading is successful, not the execution):

```
#include <TFT_eSPI.h>
TFT_eSPI tft = TFT_eSPI();
void setup() {
  Serial.begin(115200);
  delay(100);
  tft.init();
  tft.setRotation(0);
  tft.fillScreen(TFT_WHITE); 
  tft.setTextColor(TFT_BLACK);
  tft.setTextSize(2);
  tft.setCursor(50, 50);
  tft.print("ABC");
  delay(1000);
}
void loop() {
}
```

The problem with the above code is that it successfully compiles and uploads to the board but the board fails to boot at all, instead it seems to be stuck in a restart loop.

I have tried reinstalling multiple libraries (including Sensorlib) but I cannot figure out what I am missing. I also use a T Display S3 on which I display data/images using tft_espi without any issues. Let me know if I can provide with any more information to solve this.

**Side note:** 
- The red led (ON) keeps flashing regardless what is happening to the board. 
- I have tried holding down BOOT and then pressing RST, and then finally letting go of BOOT to put the board in "download mode"

lewisxhe, 2024-12-28:
It looks like your board variant is not selected from the esp32 main branch.

# 3.0.7
[image]

# 2.0.16
[image]

whzbngry, 2024-12-30:
Hi Lewis,

Thanks for your quick response.

I'm hoping that by board variant you mean Tools --> Board --> esp32 --> ESP32S3 Dev Module, then yes I have that selected.

Tried the same sketch as you:
[image]
[image]
Also tried with Sensorlib 0.2.5, same result.

Showing the example code shared in my last post being uploaded:
[image]

lewisxhe, 2024-12-31:
Please see the screenshot, 2.0.16 has the setBufferSize method. You can delete all the esp32 boards you have installed, then reinstall them and test again.
[image]

whzbngry, 2025-01-02:
I believe the issue was with my Arduino environment. I haven't reinstalled esp32 boards but everything appears to be working and compiling as it should now.
Appreciate the help!

closed 2025-01-02 by whzbngry