Xinyuan-LilyGO/LilyGo-AMOLED-Series #79: Slow framerate with new AMOLED touch 1.91 Plus

vpsmedeiros, 2024-12-29:
I just got the new AMOLED touch 1.91 Plus and the factory example shipped on it had very slow frame rate.  I loaded my program and it is still slow.   I have 3 other units prior to the plus model and they work fine.  Would you have a suggestion on what I might  do to fix this?

lewisxhe, 2024-12-29:
Plus the driver type is SPI instead of QSPI so it is normal that the frame rate is not as high as before, which is caused by the limited SPI rate.

psytraxx, 2024-12-30:
I am currently writing a rust driver for this SPI display and I experience the same slow framerate - I got much better results using DMA on the SPI bus but this produces artefacts in my case. I would be interested in a DMA example in this repository.

vpsmedeiros, 2024-12-30:
> Plus the driver type is SPI instead of QSPI so it is normal that the frame rate is not as high as before, which is caused by the limited SPI rate.

Ah okay. Thanks

closed 2025-02-13 by github-actions[bot]