Xinyuan-LilyGO/LilyGo-AMOLED-Series #80: Pin to turn off backlight for Lilygo T4 S3?

pixelwave, 2024-12-30:
I am using ESPHome with LVGL. Now I want to turn the screen / backlight off after a certain amount of time without interaction.

There is a [code example](https://esphome.io/cookbook/lvgl#turn-off-screen-when-idle) but I need the correct PIN to turn off the backlight:

```
# Define Backlight as Switch to turn display on/off
output:
  #Backlight pwm section
  - platform: ledc
    pin:
      number: GPIO1
      allow_other_uses: true  
    id: backlight_pwm
    #inverted: true
       
light:
  - platform: monochromatic
    output: backlight_pwm
    name: "Backlight"
    id: display_backlight
    restore_mode: ALWAYS_ON
    internal: false
```

Is there a backlight pin for the T4 S3 display I can use?

lewisxhe, 2024-12-31:
In this [issue](https://github.com/Xinyuan-LilyGO/LilyGo-AMOLED-Series/issues/70), there is a method to turn off the screen display. AMOLED has no backlight. AMOLED is self-luminous, which is different from TFT.

pixelwave, 2024-12-31:
> In this [issue](https://github.com/Xinyuan-LilyGO/LilyGo-AMOLED-Series/issues/70), there is a method to turn off the screen display. AMOLED has no backlight. AMOLED is self-luminous, which is different from TFT.

Thanks for clarifying @lewisxhe. I did read that other issue but I do not know how to translate this in ESPHome/YAML language. 

Anybody did this before for AMOLED displays?

lewisxhe, 2024-12-31:
I don't use esphome, you can go to the esphome page and submit a feature request to them

pixelwave, 2024-12-31:
I guess very unlikely for them to care right now about specific hardware like this. Unfortunately.

The way I solved it for now is to bring up a blank page after timeout and stop LVGL:

```
touchscreen:
...
    on_release:
      - if:
          condition: lvgl.is_paused
          then:
            - logger.log: "LVGL resuming"
            - lvgl.resume:
            - lvgl.widget.redraw:
            - lvgl.widget.hide: p_amoled_off # AMOLED
```

```
lvgl:
  on_idle:
    timeout: !lambda "return (id(display_timeout).state * 1000);"
    then:
      - logger.log: "LVGL is idle"
      - lvgl.widget.show: p_amoled_off # AMOLED
      - delay: 1s  # AMOLED
      - lvgl.pause:

  # top layer screen appear above all other GUI
  top_layer:
    widgets:
      # blank screen for AMOLED displays
      - obj:
          id: p_amoled_off
          x: 0
          y: 0
          width: 100%
          height: 100%
          bg_color: 0x000000
          bg_opa: COVER
          radius: 0
          pad_all: 0
          border_width: 0
          hidden: true
```

```
# set inactivity duration until LVGL / screen turns off
number:
  - platform: template
    name: Screen timeout (LVGL)
    optimistic: true
    id: display_timeout
    unit_of_measurement: "s"
    initial_value: 30
    restore_value: true
    min_value: 5
    max_value: 1800
    step: 1
    mode: box
```

lewisxhe, 2024-12-31:
Displaying black is the same as turning off the screen. Black means turning off the screen by default.

pixelwave, 2024-12-31:
> Displaying black is the same as turning off the screen. Black means turning off the screen by default.

Yes. Only downside is probably the still higher power consumption vs. the "disp_sleep" variant. But for now and my current projects it is sufficient. 

Maybe there will be an elegant solution later ...

closed 2025-02-14 by github-actions[bot]

pixelwave, 2025-02-14:
The correct solution was actually really simple in ESPHome/LVGL there is a brightness value you can turn to 0:

```
display:
  - platform: qspi_dbi
    id: display_amoled
    model: RM690B0
    data_rate: 80MHz
    spi_mode: mode0
    dimensions:
      width: 450
      height: 600
      offset_width: 16
    color_order: rgb
    invert_colors: false
    brightness: 255
    cs_pin: 11
    reset_pin: 13
    enable_pin: 9
    update_interval: never
    auto_clear_enabled: false
```