Xinyuan-LilyGO/LilyGo-AMOLED-Series #81: T4-S3 firmware upload

jgauchia, 2024-12-31:
Hi, I'm trying to upload firmware for T4-S3 (compiled) but doesn't works, always monitor return bad header address. I'm trying to add support for other libraries like lovyangfx, etc... but i need to verify that boards works well with default firmware supplied.

There are another one?

Thank you

lewisxhe, 2024-12-31:
`bad header address` You need to check whether you have selected the options in the README
[image]

jgauchia, 2024-12-31:
I'm using web util from espressif in order to upload firmware.bin provided in your git. Isn't a problem with board definition, simple "hello world" in Pio works without problems

jgauchia, 2024-12-31:
Only that i need is upload default firmware provided in board when purchase it

lewisxhe, 2024-12-31:
https://github.com/Xinyuan-LilyGO/LilyGo-AMOLED-Series/tree/master/firmware

jgauchia, 2024-12-31:
Yes, I know, but doesn't works... or maybe something is wrong in your guide

jgauchia, 2024-12-31:
```
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x7 (TG0WDT_SYS_RST),boot:0x29 (SPI_FAST_FLASH_BOOT)
Saved PC:0x40043ac8
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce3808,len:0x4bc
load:0x403c9700,len:0xbd8
load:0x6c746f6f,len:0x6564616f
Invalid image block, can't boot.
ets_main.c 329 ```

Raise this error....

lewisxhe, 2025-01-02:
There is no problem with the firmware test, the link is the recorded video.
Maybe it's your step that has a problem, not me
[youtube](https://youtu.be/VTZApu81040)

jgauchia, 2025-01-02:
> There is no problem with the firmware test, the link is the recorded video.
> Maybe it's your step that has a problem, not me
> [youtube](https://youtu.be/VTZApu81040)

Ok, sorry I see the mistake, It was default start address 😔. Thank you I'll try

jgauchia, 2025-01-02:
Solved

closed 2025-01-02 by jgauchia