Xinyuan-LilyGO/LilyGo-AMOLED-Series #82: T3- Dispay Amoled Touch, Power from pin 5V

zorro1011, 2025-01-03:
HI and happy new year to all

I read another post where we talk about Pin 5v's power supply and I did not understand the serial thing and I am very confused and I would not like to do mess.

in my code, I have println which I used in the tests, and I do not use when the "T-SPLAY S3 AMOLED Touch" I use it normally.
In addition, I am using the serial (Gpio43 and 44) ​​for the Sim800L module.

I can use the 5V pin to power the card instead of the USB-C connector?
which of the 2 pin 5V ?
(I'm not using the BAT connector)

thanks
best regards

lewisxhe, 2025-01-06:
VBUS and USB-C are the same line. It is best to use an external power supply to power the SIM800L. VBUS is also an option because its current demand is very large.
[image]

zorro1011, 2025-01-06:
many thanks

but where do you get the diagrams that you post in the answers?
is there a global diagram that I can consult too?

lewisxhe, 2025-01-06:
https://github.com/Xinyuan-LilyGO/LilyGo-AMOLED-Series/blob/master/schematic/T-Display-S3-AMOLED-Touch.pdf

zorro1011, 2025-01-07:
wow, thank you so much

closed 2025-02-21 by github-actions[bot]