Xinyuan-LilyGO/LilyGo-AMOLED-Series #83: Disable Charge Status Light in ESPHome

carbuthn, 2025-01-03:
I using a T4 S3 Amoled display, I program using ESPHome. I need to disable the charge status light. I did it with Arduino code from examples, but it comes back. You would a lot more displays if you provided support inside ESPHome. This applies to T3 S3 Amoled and T-Display S3 Pro also.

lewisxhe, 2025-01-06:
I don't use ESPHome, but this thread has discussed it
https://github.com/Xinyuan-LilyGO/T-Display-S3-Long/issues/31

carbuthn, 2025-01-07:
i2c: # Add an Id to the touchscreen I2C bus
  sda: 5
  scl: 6
  scan: false
  id: i2c_bus

sy6970:
  i2c_id: i2c_bus
  state_led_enable: false

external_components:
  - source: github://buglloc/esphome-components
    components: [ sy6970 ]

# This works for the T4 S3 Amoled and the T-Display S3 Pro.

closed 2025-02-21 by github-actions[bot]