Xinyuan-LilyGO/LilyGo-AMOLED-Series #87: Correct GPIO pins for T3-S3-Amoled Plus

carbuthn, 2025-01-14:
I am having problems get the correct pins for the QSPI and Display. I set per the picture showing the pins but display remains blank.
spi:
  id: quad_spi
  type: quad
  clk_pin: 47
  data_pins: [18, 7, 48, 5]

i2c:
  sda: 3
  scl: 2

display:
    cs_pin: 6
    reset_pin: 17
    enable_pin: 38

I am using ESPHome and have several of the Non-Plus versions that work correctly.

Do you have a list that shows the correct pins for the Amoled Plus?

lewisxhe, 2025-01-15:
Amoled Plus is an SPI interface, not QSPI.
[image]

carbuthn, 2025-01-15:
That is the picture that I mentioned, it does not work with either of the two T-display Amoled Pluses that I have. The description says that it uses a QSPI but the picture above does not say what the four data pins are. The picture is similar to the regular Amoled display (non-plus) and those work with the pins in my original post. I also tried setting the enable to pin 9 instead of 38 but that did not work.

lewisxhe, 2025-01-15:
amoled plus is a driver for the SPI interface, not QSPI. QSPI is only 1.91 amoled, not amoled plus.

esphome currently does not seem to have an adapter for amoled plus

carbuthn, 2025-01-17:
Thanks I will change the code that I was using to see if I can get it to work with the SPI Interface.

It may help to change the description page that currently says QSPI for the Amoled plus.

[image]

lewisxhe, 2025-01-20:
Thanks, I will report this to sales

closed 2025-03-06 by github-actions[bot]