Xinyuan-LilyGO/LilyGo-AMOLED-Series #89: DMA for pushColors

kennygarreau, 2025-02-05:
I've been trying to write a DMA-enabled pushColors function for the T4 S3. I'm trying to get this working to see if I can improve the performance at all from the default pushColors, as the more BLE packets my code processes, the higher my CPU use goes (upwards of 98%) and the lower my FPS goes. My lv_task_handler is set to tick every 10ms, and the in-code timers I've setup show lv_task_handler taking 40-45ms.

The DMA transfer appears to work, and I've setup some debugging in the function. The first few lines of my Serial output are the core logic, and then you'll see the logging I've setup in the function linked below. But the problem I have is that by using this approach, the screen is completely blank. Am I missing something, or will DMA simply not work using the RM690B0? Thanks in advance!

```
 == blink k == blink front ==
SilentRide requesting mute
SilentRide requesting mute
respAlertData: INDX:1 FREQ_GHZ:10.525000 FSTR:1 RSTR:1 BAND:X BDIR:REAR PRIO:0 JUNK:0 SPEED: 0.103570
respAlertData: INDX:2 FREQ_GHZ:24.150000 FSTR:3 RSTR:1 BAND:K BDIR:FRONT PRIO:0 JUNK:0 SPEED: 0.103570
respAlertData: INDX:3 FREQ_GHZ:34.700001 FSTR:7 RSTR:1 BAND:KA BDIR:FRONT PRIO:1 JUNK:0 SPEED: 0.103570
Available DMA memory: 216932 bytes
Transaction completed, 8112 bytes transferred.
DMA buffer freed and SPI transaction complete.
Available DMA memory: 216932 bytes
Transaction completed, 16384 bytes transferred.
Transaction completed, 14416 bytes transferred.
DMA buffer freed and SPI transaction complete.
Available DMA memory: 216932 bytes
Transaction completed, 16384 bytes transferred.
Transaction completed, 12800 bytes transferred.
DMA buffer freed and SPI transaction complete.
Available DMA memory: 216932 bytes
```

```c++
void LilyGo_AMOLED::pushColorsDMA(uint16_t x, uint16_t y, uint16_t width, uint16_t height, uint16_t *data) {
    if (!spi) {
        Serial.println("SPI device not initialized!");
        return;
    }

    size_t totalPixels = width * height;
    size_t totalBytes = totalPixels * sizeof(uint16_t);
    size_t maxTransferSize = SEND_BUF_SIZE; // DMA chunk size
    uint16_t *p = data;

    setCS();

    setAddrWindow(x, y, x + width - 1, y + height - 1);
    writeCommand(LCD_CMD_RAMWR, NULL, 0);

    size_t freeMem = heap_caps_get_free_size(MALLOC_CAP_DMA);
    Serial.printf("Available DMA memory: %d bytes\n", freeMem);

    uint16_t *dmaBuffer = (uint16_t *)heap_caps_malloc(maxTransferSize, MALLOC_CAP_DMA | MALLOC_CAP_32BIT);
    if (!dmaBuffer) {
        Serial.println("DMA Buffer allocation failed!");
        clrCS();
        return;
    }

    size_t remainingBytes = totalBytes;
    spi_transaction_t trans = {0};

    while (remainingBytes > 0) {
        size_t chunkSize = (remainingBytes > maxTransferSize) ? maxTransferSize : remainingBytes;
        memcpy(dmaBuffer, p, chunkSize);

        memset(&trans, 0, sizeof(spi_transaction_t));
        trans.flags = SPI_TRANS_MODE_QIO;
        trans.tx_buffer = dmaBuffer;
        trans.length = chunkSize * 8;

        esp_err_t ret = spi_device_queue_trans(spi, &trans, portMAX_DELAY);
        if (ret != ESP_OK) {
            Serial.printf("Failed to queue SPI transaction: %d\n", ret);
            break;
        }

        spi_transaction_t *done_trans;
        ret = spi_device_get_trans_result(spi, &done_trans, portMAX_DELAY);
        if (ret != ESP_OK) {
            Serial.printf("Error in getting transaction result: %d\n", ret);
            break;
        }

        Serial.printf("Transaction completed, %zu bytes transferred.\n", chunkSize);
        remainingBytes -= chunkSize;
        p += chunkSize / sizeof(uint16_t);
    }

    heap_caps_free(dmaBuffer);
    clrCS();
    Serial.println("DMA buffer freed and SPI transaction complete.");
}
```

lewisxhe, 2025-02-06:
I haven't studied DMA transfer, but from the official examples, it doesn't seem like these few lines can make DMA effective. As far as I know, when DMA transfer is in progress, if the queue is full, the data packet will not enter the transmission, but a protection error will occur. When you break, the data is not actually written to the screen, and you should not call lv_disp_flush_reday() at this time, but need to continue waiting. I suggest you ask a question on esp32-core.

kennygarreau, 2025-02-09:
Thanks! I'll close this out and if I can get this working as intended I'll get a commit in to you. Cheers!

closed 2025-02-09 by kennygarreau