Xinyuan-LilyGO/LilyGo-AMOLED-Series #91: Lilygo T4 S3 - Battery State?

pixelwave, 2025-02-23:
Is there a way for the **Lilygo T4 S3** to read out the battery charge (in %) of a connected 3.7V lipo battery **without additional circuitry**?

Does it maybe have a built-in voltage divider for that use?

lewisxhe, 2025-02-24:
No resistor divider

pixelwave, 2025-02-24:
> No resistor divider
> 
> Read voltage See example https://github.com/Xinyuan-LilyGO/LilyGo-AMOLED-Series/tree/master/examples/PPM_Example_for_T4S3

Thanks for clarifying @lewisxhe!

Now my project is written in ESPHome. The way I understand it is that the LilyGo T4 S3 uses the AXP2101 (or a similar PMIC) to manage power, including battery charging and voltage measurement. The PMIC communicates over I2C, and the LilyGo_AMOLED library accesses the battery voltage via the SY class (likely an abstraction for the AXP2101).

To translate this to ESPHome I2C Setup first, I need to ensure I2C is configured in my ESPHome YAML file:

- for the LilyGo T4 S3, the I2C pins are typically:

```
SDA: GPIO21 (default)
SCL: GPIO22 (default)
```

**-> Are those correct?**

So my ESPHome config (yaml) would look something like:

```
i2c:
  sda: 21
  scl: 22
  scan: true  # Optional: Scans for I2C devices to confirm the address
```

And then since ESPHome doesn’t have a native component for the AXP2101, I will need to create a custom sensor using the i2c component and read the battery voltage register:

```
i2c:
  sda: 21
  scl: 22
  scan: true

sensor:
  - platform: i2c
    id: axp2101_battery_voltage
    address: 0x34  # Replace with the actual I2C address of AXP2101
    register: 0x78  # Replace with the correct battery voltage register (check datasheet)
    value_type: U_INT16  # Adjust based on the register size (e.g., 12-bit or 16-bit)
    filters:
      - lambda: |-
          return (x * 1.1) / 1000.0;  # Scale raw value (e.g., 1.1 mV per LSB) to volts
    unit_of_measurement: "V"
    accuracy_decimals: 2
    name: "Battery Voltage"
    update_interval: 5s
```

lewisxhe, 2025-02-25:
What? T4S3 is SY6970, you have already mentioned it in this thread https://github.com/Xinyuan-LilyGO/T-Display-S3-Long/issues/31

pixelwave, 2025-02-25:
> What? T4S3 is SY6970, you have already mentioned it in this thread [Xinyuan-LilyGO/T-Display-S3-Long#31](https://github.com/Xinyuan-LilyGO/T-Display-S3-Long/issues/31)

Ah sorry my bad. But the approach should stay the same I guess? 

_"SY6970 typically communicates over I2C and has registers for reading battery voltage, similar to the AXP2101. With an I2C address (e.g., 0x34) and the battery voltage register address from the SY6970 datasheet."_

So still an I2C assignment:

```
i2c:
  sda: 21
  scl: 22
  scan: true  # Scan for I2C devices to find the SY6970 address
```

Plus a sensor: 

```
sensor:
  - platform: i2c
    id: sy6970_battery_voltage
    address: 0x34  # Replace with the actual SY6970 I2C address
    register: 0x78  # Replace with the correct battery voltage register (check datasheet)
    value_type: U_INT16  # Adjust based on register size
    filters:
      - lambda: |-
          return (x * 1.1) / 1000.0;  # Scale raw value (e.g., 1.1 mV per LSB) to volts
    unit_of_measurement: "V"
    accuracy_decimals: 2
    name: "Battery Voltage"
    update_interval: 5s
```

lewisxhe, 2025-02-25:
1. WRITE  -> REG 02H  BIT6 SET 1
2. READ   ->  REG 0EH ->VAL * 20 + 2304

closed 2025-04-11 by github-actions[bot]