Xinyuan-LilyGO/LilyGo-AMOLED-Series #92: GPIO pins for T3-S3-Amoled Plus

carbuthn, 2025-02-24:
I still can not get the display to work.

1. What does the disp_te pin do?
2. Is there a display enable pin?
3. What chip does the RTC use?

Thanks

lewisxhe, 2025-02-24:
1. TE is to prevent screen tearing pin
2. Pin https://github.com/Xinyuan-LilyGO/LilyGo-AMOLED-Series/blob/1063ebc99ef30b03f3133229c0da8486d748f9d4/src/LilyGo_AMOLED.h#L180
3. PCF8563
4. [schematic](https://github.com/Xinyuan-LilyGO/LilyGo-AMOLED-Series/blob/master/schematic/T-Display-S3-AMOLED-Plus.pdf)

carbuthn, 2025-02-24:
I still can't get my Amoled Plus to work.

1. I could not find any information on what tearing pin does or what controls it.
2. Does Disp_TE have to be used?
3. I could not find an Enable pin, like on my other displays and did not find it in the configuration code that you listed.
4. Where can I find the factory firmware for the Amoled Plus, I found it for the other Amoleds but not for the plus.

Thanks

lewisxhe, 2025-02-25:
This warehouse stores the factory firmware of amoled plus. I don't understand what you mean by emphasizing it again.

Do you mean that it doesn't work after uploading the example in this warehouse, or that the program you wrote separately doesn't work?

And I have attached the schematic above. You can directly check the schematic to determine which GPIO is connected to the screen.

[image]

carbuthn, 2025-02-25:
1. Does the factory firmware for the t3 s3 amoled and the amoled plus?
2. What should control disp_te (gpio 9). Should it be high or low logic.

lewisxhe, 2025-02-26:
1. Of course
2. te  pin can be unused, and it can still be displayed without setting it.
You can see the source code here
https://github.com/Xinyuan-LilyGO/LilyGo-AMOLED-Series/blob/1063ebc99ef30b03f3133229c0da8486d748f9d4/src/LilyGo_AMOLED.cpp#L401

closed 2025-04-11 by github-actions[bot]