Xinyuan-LilyGO/LilyGo-AMOLED-Series #93: TE pin usage for t-display s3 amoled

amir1387aht, 2025-03-02:
hi there, hope you glad
i wanted to use TE pin for my t-display amoled
first of all, i uncommented ``` {0x35, {0x00},        0x00}, //TE ON ``` (also increased RM67162_INIT_SEQUENCE_LENGHT)
then i disabled immediately ``` lv_disp_flush_ready ``` calling
and this is my current code for using te pin : 
```
void IRAM_ATTR te_isr_handler() {
    // This ISR is called when the TE pin signals a rising edge.
    // If a flush is pending (g_current_disp_drv is not NULL), complete the flush.
    if (g_current_disp_drv != NULL) {
        lv_disp_flush_ready((lv_disp_drv_t *)g_current_disp_drv);
        // Clear the pointer to prevent multiple calls.
        g_current_disp_drv = NULL;
    }
}

void App::setupDisplay()
{
    amoled.beginAMOLED_191();

    TE_PIN = amoled.boards->display.te;

    attachInterrupt(digitalPinToInterrupt(TE_PIN), te_isr_handler, FALLING);

    beginLvglHelper(amoled);
}
``` (g_current_disp_drv  is current display)

my board just show first frame and then stops, hope find answer
thanks

lewisxhe, 2025-03-03:
I have no experience with this, please refer to this thread https://github.com/Xinyuan-LilyGO/T-Display-S3-AMOLED/issues/2

And the interrupt handler should not call lv_disp_flush_ready directly, right? This is unsafe

amir1387aht, 2025-03-03:
what do you mean by unsafe?

lewisxhe, 2025-03-04:
nothing .

closed 2025-04-18 by github-actions[bot]